rtic-rs/rtic #23: Macros are not portable when supporting more than one board/cpu

hmvp, 2017-05-10:
I follow the copper book and your blog posts (great stuff!) to create a proof of concept for some STM boards we had lying around: https://github.com/hmvp/rust_poc

It seemed logical to make to modules within the poc library for board support. I used the peripherals! macro in both resulting in a conflict because both modules declared a GPIO peripheral with the same name. 
I solved this by not using the macro: 
https://github.com/hmvp/rust_poc/blob/master/src/stm32f3348_discovery/mod.rs
 
Uncommenting the macro stuff gives the error.

For the task! macro I did not even try to move the common stuff to the bsp modules but I suspect I will get similar issues.

NB: I am quite new to this so I might have missed an obvious way to to things that does not lead to issues..

NB2: the two binaries now are named the same as the board but ideally the nucleo one should be able to run on both boards (if the cpu specific stuff was moved to the bsp module)

hmvp, 2017-05-16:
Yes  I see how a different structure will lead to this not happening. In this case I wanted to move as much as possible to the BSP modules since the application was way smaller than the 'boilerplate' to set it up. I maybe should not have used the rtfm crate since I had only one task/function. But this was what the examples said and so I didn't try without rtfm.
Keeping it in the application crate should not be a problem for larger applications.

Do note that the peripherals I defined did not conflict. They used different 'svd crates' eg:

```
peripherals!(stm32f3x4, {
    GPIOA: Peripheral {
        register_block: Gpioa,
        ceiling: C0,
    },
}
```
and 
```
peripherals!(stm32l4x6, {
    GPIOA: Peripheral {
        register_block: Gpioa,
        ceiling: C0,
    },
}
```
in the same crate will conflict even though they are for different cpus an will never be built into the same binary. Thus you cannot create a crate that builds the same app for two cpus. You always have to split it out into different crates.

Changing the macro to:
```
#[macro_export]
macro_rules! peripherals {
    ($device:ident, {
        $($PERIPHERAL:ident: Peripheral {
            register_block: $RegisterBlock:ident,
            ceiling: $C:ident,
        },)+
    }) => {
        $(
            #[allow(private_no_mangle_statics)]
            #[no_mangle]
            static $device_$PERIPHERAL:
                $crate::Peripheral<::$device::$RegisterBlock, $crate::$C> =
                    unsafe { $crate::Peripheral::_new(::$device::$PERIPHERAL) };
           // To be able to use the non prefixed name within this file
           use $device_$PERIPHERAL as $PERIPHERAL;
        )+
    }
}
``` 
(note the parts with $device_$PERIPHERAL)
might work and still be save (not tested, not even sure if it compiles)

TeXitoi, 2019-02-12:
This issue looks really outdated. @japaric closing it?

closed 2019-02-13 by japaric