rtic-rs/rtic #25: [RFC] drop the threshold token from tasks' signature

japaric, 2017-05-16:
Tasks always start with a threshold level that matches the priority level of the task. It's safe to add a method to the priority token to synthesize a threshold token (at the level) from it:

``` rust
impl Priority<N> {
    pub fn threshold(&self) -> Threshold<N> { .. }
}
```

then the threshold token could be dropped from tasks' signatures making them simpler:

``` rust
fn t1(task: Exti0, ref prio: P1) {
    let thr = &prio.threshold();
    let r1 = R1.access(prio, thr);
}
```

Should we do this?

- Does this change make things more complicated or simpler for newcomers?
- This is a breaking change.
- It's almost always the case that you'll have to call the `threshold` method in a task. Would that feel repetitive?
- Should `threshold` be named differently? Perhaps `base_threshold`?

etrombly, 2017-05-25:
I think if you will have to call the threshold method anyway this isn't really simplifying anything. The current way has been fine for me so far. The only way I'd see this being really useful is if you add support for passing in resources to the task, then maybe having a shorter signature would be nice.

japaric, 2017-07-29:
With the changes in #34. The threshold token is now mandatory (it's needed to call generic functions that use the `Resource` trait) but all the other tokens are gone.

closed 2017-07-29 by japaric