rtic-rs/rtic #27: Add support for exceptions

idubrov, 2017-05-24:
I don't know how much value would be to support *all* possible exceptions (as tasks), but at least `sys_tick` should be supported, I think.

japaric, 2017-05-24:
Yeah, we definitively want to do this, but there are some complications with the implementation and changes around the user interface, IIRC. I think exceptions would have a different syntax in the tasks! macro; exceptions in tasks! won't work by default because you have to disable the "exceptions" feature of the cortex-m-rt crate; the mechanism to override the other exceptions, the ones not used as tasks, would change (it won't use the `EXCEPTIONS` variable anymore). At least that's the expected breakage; I haven't tried to implement this yet.

japaric, 2017-06-12:
Exceptions that have configurable priority:

- UsageFault (6)
- BusFault (5)
- MenManage (4)
- SVCall (11)
- SysTick (15)
- PendSV (14)

[Source](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0553a/CIAGECDD.html).

These are candidates for being tasks. Their priorities are set through the System Handler Priority Registers. The number in parenthesis in the position of the exception priority in those registers.

japaric, 2017-06-28:
#31 would fix this.

japaric, 2017-07-29:
Done in #34. Only the system handlers (PENDSV, SYS_TICK and SVCALL) can be used as tasks.

closed 2017-07-29 by japaric