rtic-rs/rtic #29: Weird behavior with trivial 'idle' function

idubrov, 2017-05-26:
I'm observing weird behavior if I have a very trivial idle function (just an infinite loop): "init" function gets called twice.

What I observe is that:

1) Second time init is called ::cortex_m::register::primask::read().is_active() returns true (which means interrupts are enabled, as I understand).
2) If "idle" is trivial, I cannot set breakpoint to it (optimized away by compiler?)
3) If "idle" is not trivial (loop with NOP or hprintln! at the beginning), everything works as expected.

I inserted breakpoint just before the "idle(p0, t0);" invocation and here is what I see in GDB:

  0x80007a4 <x2_feed::main+8>                     bkpt   0x0000 ; Breakpoint I inserted just before the "idle" call
  ; Hey, where is my infinite loop?
  ; I think this piece of code is "closure" that calls "init'... That kind of explains it                                                                                                                                             
  0x80007a6 <x2_feed::main::{{closure}}>          push   {r7, lr}                                                                                                                                        
  0x80007a8 <x2_feed::main::{{closure}}+2>        mov    r7, sp                                                                                                                                          
  0x80007aa <x2_feed::main::{{closure}}+4>        movw   r0, #8594       ; 0x2192

(this is probably not RTFM issue per se)

japaric, 2017-05-26:
> I'm observing weird behavior if I have a very trivial idle function (just an infinite loop)

It may be a known LLVM misoptimization where LLVM replaces the function call with an "undef" value in LLVM IR. Does the problem goes away if you add impurity to the idle function? Something like ptr::read_volatile.

japaric, 2017-05-26:
See rust-lang/rust#39253 and rust-lang/rust#18785.

idubrov, 2017-05-26:
Yeah, it looks like it's what it is. Thanks!

closed 2017-05-26 by idubrov