rtic-rs/rtic #32: Experiment: timing semantics

japaric, 2017-06-29:
The original RTFM language had timing semantics baked into it. It was possible
to e.g. schedule task A to run N milliseconds after task B started where A and B
could even be the same task.

This issue will be used to track progress on incrementally reproducing this
functionality in cortex-m-rtfm.

# Requirements

The implementation must only use core peripherals and must not require dynamic
allocations.

# Milestones

- [ ] recurrency: Have a single task execute periodically e.g. every second.
  "Task A must run N milliseconds after task A started"

- [ ] multiple: Have two tasks with different priorities execute periodically at
  different frequencies.

- [ ] dispatcher: Same as before but the tasks will run at the same priority.
  This time both tasks must be dispatched from a *single* interrupt.

- [ ] offset: Have a task A execute periodically, and a task B execute after N
  milliseconds from the start of task A.

- [ ] conditional: Same as before but have task B only be scheduled under some
  condition: for example every other execution of the task A.

- [ ] srp: Make sure all tasks have their own resources and local data and that
  they adhere to SRP semantics.

# Implementation

The implementation will live in a feature branch for the duration of the
experiment. The implementation should use DWT.CYCCNT as a monotonic timer and
SYS_TICK to generate timeouts interrupts.

bergus, 2018-04-21:
I guess this is what what announced in the [Outro of the *Fearless concurrency in your microcontroller* post](http://blog.japaric.io/fearless-concurrency/#outro)?

> There’s a bunch of stuff in the pipeline: […] Offset based timing semantics backed by a device agnostic implementation, offering task chaining with periodic as well as arbitrary timing patterns. Leaving all device specific timers free to use for application purposes.

I am really looking forward for this and would like to help if you think this is a suitable task for a Rust beginner like me. Do you have any particular architectural ideas for this already?

Until this feature is completed, I suppose the way to go is to put my own timed event loop in the `idle` function?

japaric, 2018-04-21:
I'm working on this and #33 right now. I hope to have a MVP ready by next week.

> I am really looking forward for this and would like to help if you think this is a suitable task for a Rust beginner like me.

It's best to wait for the MVP and then provide feedback about the user interface.

> Do you have any particular architectural ideas for this already?

The implementation is very similar to the one described in [this paper](http://ltu.diva-portal.org/smash/record.jsf?pid=diva2%3A1013030&dswid=-6895) but allows for payloads.

japaric, 2018-04-29:
@bergus the "tq" branch has a functional implementation of this and #33 if you want to give it a try *but* it has zero documentation at the moment; only examples.

bergus, 2018-05-07:
@japaric Reading the code is very educational, thanks :-) I think I'll have some feedback on implementation and usability later this week.

japaric, 2018-05-08:
@bergus we are currently working on an mdbook that contains both a user guide ("how do I use this?") and a high level overview of the implementation ("how does this actually work?"). You can find it at https://japaric.github.io/cortex-m-rtfm/; it's still WIP and being updated every now and then. You'll see that the API used in the book is different than the one used in the examples of the `tq` branch; that's because we are still iterating on the terminology and the API, but we'll probably end up using something closer to what's being used in the mdbook.

bergus, 2018-05-08:
Thanks, I guess I already could extract the main ideas from the code :-) I got a few questions though:

* How much overhead do the `Context`s and `Resource`s have, are they zero-sized? I hope they get inlined as constant pointers to the static variables?
* Is it possible to run tasks with `priority: 0` in the idle loop?

commit c631049e (2018-11-03): v0.4.0

closes #32
closes #33
commit 777765e5 (2018-11-03): Merge #97

97: v0.4.0 r=japaric a=japaric

closes #32
closes #33
PR #97 (2018-11-03): v0.4.0
closes #32
closes #33
closed 2018-11-03 by bors[bot]