rtic-rs/rtic #35: RFC: Provide access to read-only registers without asserting data-race freedom

Emilgardis, 2017-07-18:
# Summary
Provide a way to access read-only registers without using unsafe code.

# Motivation
Currently, `rtfm` provides methods to access peripherals in a concurrent way.
This means that even when we know a read is safe (and doesn't change the value), rtfm doesn't know because there is no way for us to tell that a register is read-only (and safe).

### Example

This example shows one way how to do this in rtfm v0.1.1
(It is rather dumb in the way it is setup, but it was the simplest way I could demonstrate this)

A button is setup on PA1, another button is on PA2 and a led on PA5.
The button on PA1 controls the led on PA5 inside the idle loop. The button on PA2 is setup to trigger an exti interrupt that controls the led on PA5.

```rust
extern crate cortex_m;
extern crate cortex_m_rt;
#[macro_use]
extern crate cortex_m_rtfm;
extern crate pac; // Our peripheral access crate

use rtfm::{C1, P0, P1, Resource, T0, T1, TMax};

peripherals! {
    GPIOA: Peripheral {
        register_block: GPIOA,
        ceiling: C1,
    },
    EXTI: Peripheral {
       register_block: EXTI,
       ceiling: C1,
    },
}

fn init(prio: T0, thres: &TMax) {
    // setup exti on PA2 ...
}

fn exti_pa2(mut task: EXTI0, ref prio: P1, ref thres: T1) {
    let exti = EXTI.access(prio, thres);
    let gpioa = GPIOA.access(prio, thres);
    
    // Clear pending exti and change led PA5
}

fn idle(ref prio: T0, ref thres: T0) -> ! {
    let gpioa_ro = pac::GPIOA.get();
    loop {
        if unsafe { (*gpioa_ro).idr.read().idr1().is_set() } {
            thres.raise(&GPIOA, |threshold: &T1| {
                let gpioa = GPIOA.access(prio, threshold);
                // Change state of led on PA5
            }
        }
    }
}

```

# Design

With information on read-only registers, rtfm should enable us to write something like the following.

```rust
registers_ro! {
        GPIOA: [IDR,],
}

fn idle(ref prio: T0, ref thres: T0) -> ! {
    let gpioa_idr = GPIOA.access_read::<GPIOA_IDR>();
    loop {
        if gpio_idr.read().idr1().is_set() {
            thres.raise(&GPIOA, |threshold: &T1| {
                let gpioa = GPIOA.access(prio, threshold);
                // Change state of led on PA5
            }
        }
    }
}
```
[gist of codes](https://gist.github.com/Emilgardis/ca36823f1bb591d4637ff438cb8902b0)

Here, the `register_ro` macro provides an static object called `GPIOA_IDR` that is `bare-metal::Register<RW> where RW: cortex-m::Read + !cortex-m::Write`.

The function `Peripheral::access_read<T> where T: _` returns `T`.

## Drawbacks & Questions
- This seems to require a big change in the current ecosystem for such a small improvement.
- Is it possible to combine this with the ideas coming from the [embedded-hal] project?
  Can you generalize over this?
- Are all reads atomic?

japaric, 2017-07-18:
@Emilgardis Thanks for starting this discussion. I think there are two topics,
that I think could be addressed in an orthogonal fashion, that this RFC is
touching:

- Resource protection at the register level. Currently RTFM provides protection
  at the peripheral level, and

- Adding read-only, read/write, write-only semantics to the Resource
  abstraction.

Let me use this chance to write down my thoughts about these two topics before
commenting on your RFC.

NOTE: Everything written below is assuming RTFM v2 (cf. #34)

# Register granularity

I think this is something we definitively want to do at some point because it
would let us futher reduce the number of critical sections in some applications.
However, this is going to be tricky to implement.

The main challenge I see with registers as resources is that within a peripheral
some registers are ~~quantum entangled~~ implicitly connected. By this I mean
that modifying one register will modify the other. One example of this are the
BSRR (Bit Set / Reset Register) and ODR (Output Data Register) registers in
STM32 chips. BSRR is write only and ODR is read write but writing to BSRR
changes the state of ODR. If single registers can be used as resources then
this implicit connection can lead to data races:

``` rust
// priority = 1
fn t1(r: T1::Resources) {
    loop {
        // non atomic RMW operation
        r.GPIOA_ODR += 1;
    }
}

// priority = 2
fn t2(r: T2::Resources) {
    loop {
        // sets all the pins
        r.GPIOA_BSRR = 0x0000_FFFF;

        // clears all the pins
        r.GPIOA_BSRR = 0xFFFF_0000;
    }
}
```

In this a data race? Yes, by definition. Is it harmful? It depends on the actual
application (this example is rather contrived). As registers are word sized data
races on them won't cause torn reads but can affect an application behavior in
non deterministic ways.

The solution to this particular issue is simple to write down: "consider both
BSRR and ODR as a single resource" but hard to implement without human
intervention: how can a tool possibly known that ODR and BSRR are related? SVD
files don't contain this information.

My other gripe with register level protection is that it may actually not be
fine grained enough. An example of this is the Interrupt Status Register (ISR)
and Interrupt Flag Clear Register (IFCR) pair found in the DMA peripheral of
STM32 devices. ISR is read only and IFCR is write only. Let's assume that we
consider them both a single resource (a write to IFCR changes ISR). Each
register has a 4-bit field per DMA channel. With register granularity if you
need one task to work with channel 1 and another one to work with channel 2 and
the tasks have different priorities then you are forced to use critical sections
even though the bit fields of each channel are decoupled:

[... 113 lines cut ...]
> improvement.

If we want to tackle these two problems then I think we should think about the
two *general* problems; breaking the ecosystem to solve these two general
problems would be more worthwhile than to solve a particular case, IMO.

> Is it possible to combine this with the ideas coming from the [embedded-hal]
> project?

I think so. Since the HAL traits abstract away the struct it doesn't matter if
the implementer of e.g. `Serial::Write` is `&USART1` or `(&USART1_SR,
&mut USART1_DR)`.

This made me realize that if we want to work with register instances then we
would need to distinguish e.g. between the SR of USART1 and the SR of USART2 --
they have to be different types -- otherwise a `struct Serial { sr: &SR, dr:
&mut DR }` could be created with the `SR` of USART1 and the DR of USART2;
clearly that's not something we want to allow. Or we could address this mismatch
in some configuration layer (some lint) and not in the types of the application
code.

> Are all reads atomic?

If the register is 32 bit or smaller then the read operation is performed in a
single instruction.

perlindgren, 2017-07-19:
Hi

Following up on this discussion.

I believe that the real strength of RTFM is the resource protection in between software objects, (including encapsulating peripheral access). Hence, we could leave low-level access (registers/periperals) as being up the the implementer of a “driver”.

As Jorge mentions, its going to be tricky to capture the actual semantics of registers/peripheral access, the SVDs does not contain this info, and moreover is may even be application specific. E.g., a single I/O pin, could be used for more than one purpose (in multixplexed setting). This is clearly out of reach for any general protection mechanism.

Moreover, register access is in general “unsafe”, as writing to a register may cause the system to hard fault (so what we can offer in terms of protection using RTFM is to that end of little help).

Nevertheless, we can enforce some degree of “hygiene” based on the SVDs (and that is exactly what svd2rust offers), namely a “safe” (and yet efficient) way to access registers (safe in the meaning that bit masking is correct by construction).

Moreover, I believe it may be useful at application layer to have control of what registers/perpipherals are used by the application, either clear from specification (you grant only a subset of the memory map to each component in the system in a declarative manner), or you get a report at compile time of the usages. This could greatly help a programmer to spot potential collisions/race conditions, and remedy those at a higher level protecting those by Resources.

We have experimented with the concurrency model leading up to RTFM-core (a simple task/resource model) and an object oriented abstraction RTFM-cOOre. The latter uses RTFM-core as the underlying model of execution, and adds the notion of Concurrent Reactive Objects (CRO). CRO objects communicate through p2p messages (both synchronous and asynchronous).

Each message is a tuple M < b, d, O, m, p, r >
b, is the base-line (absolute time of arrival)
d, is the relative deadline (from which static priorities can be derived)
O, is the receiving object ID
m, is the receiving method ID
p, is the parameters
r, is a return value (in case of a synchronous message), packing r into the message is just an implementation detail

(Messages can be seen as Events in the system)

The Environment (I/E) hardware platform provides a set of Message/Event sources (i.e, the Interrupt vector), and a set of accessible registers. Each interrupt (message/event) source may be associated to an asynchronous message M, identifying the target object, method, (null parameters), and no return value. The base-line b, is derived either by HW timestamping (if not available an approximate arrival time can be derived through reading the DWT-CYCNT, but it will be exposed to jitter due to the scheduling).  The interrupt priority is derived from M.d.

The execution of a message amount to the run-to-completion execution of the corresponding method (m) on the object (o). Race free execution is ensured by associating each object (instance) to a Resource (instance). The ceiling of the Resource (ceil(o)) is derived from the maximum priority of all messages in the system targeting o. So in effect, objects are protected (similar to Ada protected objects), and execution amounts to “claiming” (o) for the execution of (m).

During execution a method (m) may emit asyncronous messages and synchronous messages.

Asynchronous messages:
Inherited:
async(o,m,p) -> M(b, d, m, p, ()), here (b) and (d) are inherited from the sender.

Postponed:
async(a, d, o, m, p), -> M(b+a, d, m, p, ()), where a is a base-line offset, (relating to the arrival time for the sender). This allows, e.g., periodic behaviour to be captured, by just posting yourself with a time offset. [Here we made a restriction that cyclic event chains must be triggered only from the the “reset” interrupt (else the number of active cyclic chains cannot be bound).]

Synchronous messages:
r = sync(o,m,p), -> M(b, d, m, p, r),  here (b) and (d) are inherited from the sender.

In effect, the complete chain of synchronous messages will be executed as a “task”, either stemming from an external event (interrupt) or from an internal event. Asynchronous messages forms new tasks.

Implementation of message passing is possible to implement ontop of the RTFM-kernel (keeping the actual kernel super simple, we already have it even in Rust ;)

For each priority in the system, we associate a message queue (essentially a CRO object with methods to enqueue and dequeue messages). The ceiling will be that of the highest priority task (sender) that access the queue.

For postponed messages, we may have a single (shared) message queue, a queue for each message priority, or anything in between. We will have some additional blocking with a single queue approach, but in general queue operations are quite cheap for the size of systems we target, so its probably the way to go. (For a more detailed analysis see our paper on the subject http://dl.acm.org/citation.cfm?id=2907979.)

One possible component model based on CRO has been proposed, (Concurrent Reactive Components (CRC)). This model offers a port abstraction to objects, allowing hierarchies to be defined. For more details: http://www.jcomputers.us/vol4/jcp0412-17.pdf

Ok, so how does this relate to the RFC at hand.

1. CRO objects offers the means to encapsulate state in a race free manner. (State here can be both software and hw….)
[... cut ...]
At this point I don’t have a clear answer to what is best, but in either case, I strongly believe that a generic model of peripherals is not feasible, and that a CRO/CRC abstraction makes sense. Also I have not yet seen any other programming model that deals with this problem in a sensible way (common thread based concurrency models are a big mess), so we have the opportunity here….

In any case the RTFM-kernel model is a great outset for single core scheduling, that we have to our advantage.

Looking forward to further discussions.

Best regards,
  Per

japaric, 2018-11-03:
I believe the RTFM part of this feature request is done as of v0.4.x. `static` resources can be used for "read-only" access (shared (`&-`) references only). The peripheral / register can be done writing abstractions on top of an API like the one svd2rust gives you. (svd2rust can't give you read-only APIs because SVD don't contain enough information; for example, they don't indicate if a read operation also clears some bits of the registers -- such read operation needs an exclusive (`&mut-`) reference to the register, etc).

closed 2018-11-03 by japaric