rtic-rs/rtic #36: [RFC] & / &mut access semantics for resources

japaric, 2017-07-23:
# Motivation

Currently when you assign a resource to a task the task gains exclusive access
to the resource: it can both `claim` *and* `claim_mut` the resource.

This RFC proposes extending the access model to let tasks declare whether they
need exclusive access (`&mut-`) or shared access (`&-`) to their resources. This
extra information would let the `app!` macro optimize access to resources
reducing the number of required critical sections, at least in some scenarios.

# Design

We extend, in a backward compatible fashion, the syntax around the `resources`
array. Today the syntax looks like this: `[A, B, C]`; here the task has
exclusive access to the resources `A`, `B` and `C`. The syntax will be extended
to allow specifying shared or exclusive access through the `&` and `&mut`
operators. So, `[&A, &mut B, &C]` indicates that the task has shared access to
the resources `A` and `C`, and exclusive access to the resource `B`.

Here's an example:

``` rust
app! {
    resources: {
        static COUNTER: u64 = 0;
    },

    idle: {
        resources: [&mut COUNTER],
    },

    tasks: {
        EXTI0: {
            enabled: true,
            priority: 1,
            resources: [&COUNTER],
        },

        EXTI1: {
            enabled: true,
            priority: 2,
            resources: [&COUNTER],
        },
    },
}
```

In this case *both* tasks, `EXTI0` and `EXTI1`, get lockless "read access" to
the `COUNTER` resource. In today's model `exti0` would need a critical section
(`claim`) to read `data`.

``` rust
task!(EXTI0, exti0);

fn exti0(_t: &mut Threshold, r: EXTI0::Resources) {
    let data: &u64 = r.COUNTER;
}

task!(EXTI1, exti1);

fn exti1(_t: &mut Threshold, r: EXTI1::Resources) {
    let data: &u64 = r.COUNTER;
}
```

Whereas `idle`, which has lower priority than both tasks, also gets lockless
"read access" to the resource but requires a critical section (`claim_mut`) to
mutate the data.

``` rust
fn idle(t: &mut Threshold, r: idle::Resources) -> ! {
    loop {
        {
            let data: &u64 = r.COUNTER.deref();
        }

        r.COUNTER.claim_mut(t, |counter, _t| *counter += 1);

        {
            let data: &u64 = r.COUNTER.deref();
        }
    }
}
```

This new access model imposes a new constraint on the resource data: it must now
implement the `Sync` trait. This forbids interior mutability (e.g. `Cell` /
`RefCell`) which could lead to data races.

If you don't want the `Sync` constraint on your resource data you can continue
to use the old syntax with no access (`&-` or `&mut-`) information.

# Possible applications

## Multiple reader, single writer resource

This is the example shown in the example above. Only the writer needs to lock
the resource to modify it. The writer needs to run at the lowest priority,
though, or this pattern won't reduce the number of critical sections required.

## Initialize once, then read-only

A resource shared by tasks where none needs exclusive access (`&mut-`) to it can
be "locklessly" shared (`&-`) by all them. IOW, the resource can only be "read"
by the tasks. The only place where one can mutate (`&mut-`) the resource is
in `init`.

``` rust
app! {
    resources: {
        static FREQUENCY: u32 = 0;
    },

    idle: {
        resources: [&FREQUENCY],
    }

    tasks: {
        EXTI0: {
            enabled: true,
            priority: 1,
            resources: [&FREQUENCY],
        },
    },
}

fn init(_: init::Peripherals, r: init::Resources) {
    *r.FREQUENCY = 8_000_000;
}

fn idle(r: init::Resources) {
    let f: &u32 = r.FREQUENCY;
}

task!(EXTI0, exti0);

fn exti0(_: &mut Threshold, r: EXTI0::Resources) {
    let f: &u32 = r.FREQUENCY;
}
```

## Something with atomics?

The previous examples assumed that tasks that have shared (`&-`) access to a
resource can't modify it. This is not always true; tasks can modify resources
through a shared (`&-`) reference under this new system if the mutation is
*synchronized*.

Atomic types are one of the few types that allow mutation through a shared
reference *and* implement the `Sync` trait. Given these properties you can
already use them safely in `static` variables without the help of the RTFM
system. However, combining atomic types with this new access model lets you
*restrict* which tasks have access to the atomic variables, which may be useful
to prevent logic bugs. With the current, exclusive access only, model you can't
do that because it requires you to lock the resource that contains the atomic
variable when you want to access it from the lower priority tasks.

cc @pftbest this may interest you since you are exploring atomics and lock-less queues.

perlindgren, 2017-07-23:
Hi

Looks good, but I believe the gain will not be that large in any case the fully fledge claim (actually writing the basepri) is very fast, at least for the cortex-m3/m4).
So as long as the programmers API won’t be more complicated I think its fine.

Best regards,
  Per

pftbest, 2017-07-23:
Can we lift the `Sync` restriction if only one task will access the resource? The types i need are definitely not going to be `Sync`, just `Send`.
For example:
```Rust
let (rx, tx) = channel();
```
I need to have rx in one task and tx in other task, but I can't allow more than one task to access tx or rx.

Also I'm afraid that such syntax won't be possible:
```Rust
app! {
    resources: {
        static (RX, TX): (Sender<u8>, Receiver<u8>) = channel();
    }
}
```

japaric, 2017-07-23:
> Can we lift the Sync restriction if only one task will access the resource?

Sure, sounds reasonable.

> The types i need are definitely not going to be Sync, just Send.

Good because the `Resource` abstraction needs the data to be Send.

> I need to have rx in one task and tx in other task,

RFC #37 "resources initialized at runtime" may help with that part.

japaric, 2019-01-11:
Superseded by #129

closed 2019-01-11 by japaric