rtic-rs/rtic #37: [RFC] Resources initialized at runtime

japaric, 2017-07-23:
# Motivation

In the current version of RTFM every resource must be assigned an initial value
at compile time and in const / static context. This requirement can sometimes be
too strict given the current limitations of const evaluation (no loops, no
conditionals, etc.).

In other cases one may want to initialize a resource according to the outcome of
the `init` function. For instance one may want to store the frequency of the
processor, product of the initialization routine, in a variable; in the current
system this requires the programmer to manually keep a copy of the value (as a
literal) in the `resources` section of the `app!` macro, which is error prone.

The workaround for these issues is to use an `Option` for the resource and to
select `None` as the initial value. Then a value can be assigned to the resource
resource in the `init` function. The downside of this approach is that it
requires the program to `unwrap` or `match` the `Option` value whenever they
want to access the resource data. And, of course, there's no compile time
guarantee that the programmer won't forget to actually initialize the `Option`
resource to some value in `init`; forgetting to do so would cause a `panic!` at
runtime.

# Design

We make the initial value of resources optional in the `app!` macro:

``` rust
app! {
    resources: {
        static BUFFER: [u8; 32] = [0; 32];
        static FREQUENCY: u32;
        static FOO: Vec<u8>;
    },
}
```

When the initial value of any resource is omitted the signature of `init` will
be changed to force the programmer to assign a value to these resources during
the initialization phase:

``` rust
fn init(p: init::Peripherals, r: init::Resources) -> init::ResourceValues {
    // .. initialize stuff like the heap allocator ..

    // These are the initial values of the "uninitialized" resources
    init::ResourceValues {
        FREQUENCY: frequency,
        FOO: vec![a, b, c],
    }
}
```

Apart from this change in `init` there's no difference between using resources
with different initialization strategies.

## What goes on under the hood?

Under the hood the resource without initial values will actually be `static`
variables with type equal to `Option` and initialized to `None`. However, since
we know that the resources *will* be initialized in `init` and before any task
(or idle) can run we can expose them to the user without the `Option` wrapper,
plus we can apply unsafe optimizations like `intrinsics::unreachable` under the
hood to eliminate branches and avoid `unwrap`ping.

# Possible extensions

## Optimizing the pre-main initialization of `static` variables

Since the runtime initialized resources will be represented as `static`
variables containing an `Option` they'll still be initialized before `init` by
the runtime even though those initial values won't be read. This useless
initialization can be optimized away with the help of linker script magic:

``` rust
// runtime initialized resources
#[link_section = ".uninit"]
static mut FREQUENCY: Option<u32> = None;

#[link_section = ".uninit"]
static mut FOO: Option<Vec<u8>> = None;

// "normal" resources
static mut BUFFER: [u8; 32] = [0; 32];
```

`static` variables placed in the `.uninit` linker section won't be initialized
before `init`. This may require changes in `cortex-m-rt` to ensure that `static`
variables placed in the `.uninit` section end up having a valid address in the
RAM region though. cf. japaric/cortex-m-rt#32

cc @cr1901

pftbest, 2017-07-23:
> Under the hood the resource without initial values will actually be static
variables with type equal to Option and initialized to None.

Then it's not going to be a zero cost abstraction. `Option<u32>` is actually 8 bytes in size.
Too bad that `mem::zeroed()` is not a `const fn`.

perlindgren, 2017-07-23:
Hi

I have exactly the same problem when extending this approach to Concurrent Reactive Objects. I don’t think that the problem is mem::zeroed(), but rather the crippled const fn evaluation in today’s Rust. Lets hope that is solved soon (would make life in Rust much more pleasant).  (We may even have mem::unintilized() in some cases, where the framework is guaranteed to fill in data.

I have resorted to Option just to get around user having to give excessive and stupid init values. (Allthough it carries some memory and potentially run-time OH…, it is the least bad solution for now…)

Best,
   Per

japaric, 2017-07-23:
Did `mem::size_of` become a const fn yet? If yes then we could use `static mut FOO: [u8; size_of::<T>()] = [0; size_of::<T>()];` and cast `&[u8; size_of::<T>()]` into `&T` behind the hood. And pray that nothing explodes ... as in hope that LLVM won't generate UB.

pftbest, 2017-07-23:
[u8; size_of::<T>()] may not have a correct alignment, unfortunately.

perlindgren, 2017-07-23:
> [u8; size_of::()] may not have a correct alignment, unfortunately.

Hi

This is tricky and delicate stuff. The lazy statics approach won’t help either unfortunately as far as I can tell. I think proper const fn eval in Rust is what we can hope for.

Best,
  Per

japaric, 2017-07-23:
@pftbest we can use `([u8; mem::size_of::<T>()], [T; 0])` then. See bellow:

``` rust
#![feature(const_fn)]

use std::mem;

static mut FOO: ([u8; mem::size_of::<u64>()], [u64; 0]) =
    ([0; mem::size_of::<u64>()], []);

fn main() {
    unsafe {
        println!("{}", mem::align_of_val(&FOO));
        println!("{}", mem::size_of_val(&FOO));
    }
}
```

``` console
$ cargo run
8
8
```

pftbest, 2017-07-24:
@japaric, this might work. LLVM doesn't have a strict aliasing rules like in C, so if the size and alignment match I don't see why it would fail (yet).

But still it's a very ugly hack. Maybe we can ask lang team to make mem::zeroed const, until CTFE lands?

jonas-schievink, 2017-09-03:
:+1: for this feature, it would be very useful to have!

Wouldn't it be possible to use a union for this? Something like this:

```rust
union LateResource {
    uninit: (),
    init: Foo,
}

// Resources:
static mut FOO: LateResource = LateResource { uninit: () };
```

And then after init was called:

```rust
let resource_values = init(...);
FOO = LateResource { init: resource_values.FOO };
```

pftbest, 2017-09-03:
@jonas-schievink This is a great idea, thanks!

jonas-schievink, 2017-09-03:
I'd like to give this a shot if no one is working on this yet :)

PR #43 (2017-09-22): Allow initialization of resources in `init`.
Closes #37 

TODO:
* [X] Open dependent PRs
  * [x] Get them merged
* [X] Add an example ~~(I cannot currently run any of the examples since I "only" have an stm32f401)~~
* [x] Add a cfail test ensuring that `init` cannot access uninitialized resources (and possibly other tests)
  * Figure out why I cannot run these tests
* [x] Simplify code, it's quite ugly
closed 2017-09-22 by homunkulus