rtic-rs/rtic #40: Panic handler?

HemantJoshi, 2017-07-31:
I tried to build the first (hello world) app with latest cortex-m-rtfm. However got build error "error: language item required, but not found: `panic_fmt`". Fixed the app by adding panic handler related code as per "cortex-m-quickstart" examples/panic.rs. However earlier version of cortex-m-rtfm didn't require such modification. Looked into the code of cortext-m-rtfm. I found that panic handler was part of cortex-m-rtfm earlier version. However there is no panic handler in the latest cortex-m-rtfm.
So am my missing any thing? or should panic handler be part of latest cortex-m-rtfm?

pftbest, 2017-07-31:
panic handler was moved into `cortex-m-rt` crate. You can enable it with `abort-on-panic` feature:

https://github.com/japaric/cortex-m-quickstart/blob/master/Cargo.toml#L16

HemantJoshi, 2017-07-31:
Thanks, it worked.

closed 2017-07-31 by HemantJoshi