rtic-rs/rtic #46: Use toml like syntax instead of json like in app!

tekjar, 2017-09-24:
Hi @japaric @jonas-schievink. First of all, thanks for this extraordinary work. 

What do you think of having toml like syntax in `app!`.

```
app! {
    device = blue_pill::stm32f103xx,
    
    [resources]
    static ON: bool = false;
    
    [tasks.SYS_TICK]
    path = toggle
    resources = [ON]
}
```

vs

```
app! {
    device: blue_pill::stm32f103xx,

    resources: {
        static ON: bool = false;
    },

    tasks: {
        SYS_TICK: {
            path: toggle,
            resources: [ON],
        },
    },
}
```

I'm don't know if this is possible at all but I feel the top one is much cleaner. What do you think?

jonas-schievink, 2017-09-24:
I don't feel very strongly about the syntax it uses, but it's probably a good idea to use something similar to what other crates use for their macro-languages.

japaric, 2017-10-02:
One of the reasons I chose the current syntax is that it resembles other Rust-like syntax commonly found on macros. In fact we are directly using `syn` tokentree parser.

I do agree that TOML syntax looks better in most cases ... with the exception of the `static` variable syntax: Right now is equal to Rust's but it doesn't directly map to a TOML construct. I suppose we could adapt it to match TOML; maybe something like this:

``` toml
[resource.ON]
type = "bool"
value = "false" # needs to be string to be able to use constructors like "Foo::new()"
```

but it's more verbose. Also we cannot mix Rust syntax and TOML syntax because the TOML parser is not going to like it.

I, personally, don't feel strongly about changing it at the moment.

japaric, 2018-11-03:
This no longer applies. v0.4.x is using attributes on top of plain (`rustfmt`-able) Rust code.

closed 2018-11-03 by japaric