rtic-rs/rtic #52: Don't initialize memory for late resources

japaric, 2017-11-22:
Late resources are initialized at runtime so their memory doesn't need to be initialized before main / init. However, today, late resources are placed in the `.data` section so not only they get initialized (using memcpy) but they also use up FLASH memory (because everything in `.data` has some initial value stored in FLASH).

To fix this we'll have to create a new section, let's call it `.uninit`, in cortex-m-rt and then place the `static` variables that will be used with late resources in that section. The change in this repo is minimal; here's the patch (macros/src/trans.rs):

``` diff
                 None => quote! {
                     // Resource initialized in `init`
-                    static mut #_name: #krate::UntaggedOption<#ty> = #krate::UntaggedOption { none: () };
+                    #[link_section = ".uninit"]
+                    static mut #_name: #krate::UntaggedOption<#ty> =
+                        #krate::UntaggedOption { none: () };
                 },
```

But this change requires updating cortex-m-rt, specially if japaric/cortex-m-rt#43 lands. It's probably best to wait until japaric/cortex-m-rt#43 lands.

jonas-schievink, 2017-11-22:
There is a rustc "bug" hidden in here: There's no need to put an `UntaggedOption` intialized with `none` into the `.data` section, rustc/LLVM can put it in `.bss` in all cases (or `.uninit`, of course, but that's not a standard section), which would also help in this case.

jonas-schievink, 2018-03-17:
The underlying rustc bug is https://github.com/rust-lang/rust/issues/41315, which was fixed by https://github.com/rust-lang/rust/pull/48908, so this should work now

japaric, 2018-11-03:
triage: blocked on rust-embedded/cortex-m-rt#116

korken89, 2019-12-29:
Is this fixed with having the `.uninit` section?
If so can we close this?

japaric, 2020-01-06:
Yes, this has already been implemented.

closed 2020-01-06 by japaric