rtic-rs/rtic #62: Hard fault in code generated by app! macro for debug build

nallar, 2018-01-07:
Trying to use rtfm with a blue pill board. I am having trouble with tasks. Adding a task causes a hard fault for debug builds. Release builds work.

Normal code (https://gist.github.com/nallar/49a1be1e1b6aa157dff7703b7ed61b19):
```
app! {
    device: device,

    tasks: {
        CAN1_RX0: {
            path: usb_interrupt,

            resources: [USB]
        }
    }
}
```

Expanded code:
```
#[allow(unsafe_code)]
fn main() {
    let init: fn(device::Peripherals) = init;
    rtfm::atomic(unsafe { &mut rtfm::Threshold::new(0) },
                 |_t|
                     unsafe {
                         let _late_resources =
                             init(device::Peripherals::all());
                         let nvic = &*device::NVIC.get();
                         let prio_bits = device::NVIC_PRIO_BITS;
                         let hw = ((1 << prio_bits) - 1u8) << (8 - prio_bits);
                         nvic.set_priority(device::Interrupt::CAN1_RX0, hw);
                         nvic.enable(device::Interrupt::CAN1_RX0);
                     });
    let idle: fn() -> ! = idle;
    idle();
}
```

The hard fault happens on `let hw = ((1 << prio_bits) - 1u8) << (8 - prio_bits);`. Removing that line and hardcoding a priority of 1 prevents the hardfault. In particular, (1 << prio_bits) - 1u8 seems to cause it.

```
Program received signal SIGTRAP, Trace/breakpoint trap.
cortex_m_rt::default_handler::hb779949c33da971d (ef=0x20004df0)
    at C:\Users\Luna\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-rt-0.3.8\src\lib.rs:459
459         asm::bkpt();
(gdb) bt full
#0  cortex_m_rt::default_handler::hb779949c33da971d (ef=0x20004df0)
    at C:\Users\Luna\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-rt-0.3.8\src\lib.rs:459
No locals.
#1  <signal handler called>
No symbol table info available.
#2  0x0800025a in blue_pill_usb_keyboard::main::_$u7b$$u7b$closure$u7d$$u7d$::h6df592e3e3c4b436 (_t=0x20004f77) at src\main.rs:110
        init = 0x8000301 <blue_pill_usb_keyboard::init::h7fbaa15cdf06d57f>
#3  0x0800577e in cortex_m_rtfm::atomic::h04d4c70a3b2d7412 (t=0x20004f87,
    f=...)
    at C:\Users\Luna\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-rtfm-0.2.2\src/lib.rs:110
No locals.
#4  0x080001ea in blue_pill_usb_keyboard::main::h70c85532f64f8efa ()
    at src\main.rs:102
        init = 0x8000301 <blue_pill_usb_keyboard::init::h7fbaa15cdf06d57f>
#5  0x08005730 in cortex_m_rt::lang_items::start::hb2ca1dee2fb9b063 (
    main=0x80001c3 <blue_pill_usb_keyboard::main::h70c85532f64f8efa>,
    _argc=0, _argv=0x0)
    at C:\Users\Luna\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-rt-0.3.8\src/lang_items.rs:31
No locals.
#6  0x08001286 in main ()
        cortex_m_rt::RESET_VECTOR::haccb3bf5c3f20a4b = 0x8000131 <cortex_m_rt::reset_handler::h4aae2e8c4d94cc8b>
        EXCEPTIONS = {{RUST$ENCODED$ENUM$0$None = {
              __0 = 0x80022b7 <DEFAULT_HANDLER>}}, {
            RUST$ENCODED$ENUM$0$None = {__0 = 0x80022b7 <DEFAULT_HANDLER>}}, {
            RUST$ENCODED$ENUM$0$None = {__0 = 0x80022b7 <DEFAULT_HANDLER>}}, {
            RUST$ENCODED$ENUM$0$None = {__0 = 0x80022b7 <DEFAULT_HANDLER>}}, {
            RUST$ENCODED$ENUM$0$None = {__0 = 0x80022b7 <DEFAULT_HANDLER>}}, {
            RUST$ENCODED$ENUM$0$None = {__0 = 0x0}}, {
            RUST$ENCODED$ENUM$0$None = {__0 = 0x0}}, {
            RUST$ENCODED$ENUM$0$None = {__0 = 0x0}}, {
            RUST$ENCODED$ENUM$0$None = {__0 = 0x0}}, {
            RUST$ENCODED$ENUM$0$None = {__0 = 0x80022b7 <DEFAULT_HANDLER>}}, {
            RUST$ENCODED$ENUM$0$None = {__0 = 0x80022b7 <DEFAULT_HANDLER>}}, {
            RUST$ENCODED$ENUM$0$None = {__0 = 0x0}}, {
            RUST$ENCODED$ENUM$0$None = {__0 = 0x80022b7 <DEFAULT_HANDLER>}}, {
            RUST$ENCODED$ENUM$0$None = {__0 = 0x80022b7 <DEFAULT_HANDLER>}}}
(gdb) x 0xe000ed04
0xe000ed04:     0x00000803
        (icsr register, least significant byte = 0x03 = hard fault?)
(gdb) p ef
$3 = (struct ExceptionFrame *) 0x20004df0
(gdb) x/32
0x20004df0:     0xffffffff      0x00000010      0x00000001      0x08006200
0x20004e00:     0x08006200      0x08000235      0x0800025a      0x61000000
0x20004e10:     0x40006400      0x40007400      0xe0042000      0x40004c00
0x20004e20:     0x00000010      0x20004e44      0x20004f80      0x20004f77
0x20004e30:     0x08000301      0x20004f80      0x20004e34      0x20004f77
0x20004e40:     0x00000021      0xe000ed00      0xe000edf0      0xe0001000
0x20004e50:     0xe0002000      0xe000ef30      0xe0000000      0xe000ed90
0x20004e60:     0xe000e100      0xe000ed04      0xe000e010      0xe0040000
(gdb)

```

The backtrace shows it's faulting on this instruction:

`.text:0800025a 50 fa 81 f0                      uxtab	r0, r0, r1`

I don't have enough experience to debug further.

japaric, 2018-01-07:
> uxtab r0, r0, r1

Sounds like you chose the wrong compilation target. The ARM [documentation](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0489i/CIHFJCAG.html) says:

> This 32-bit Thumb instruction is available in ARMv6T2 and above. For the ARMv7-M architecture, it are only available in an ARMv7E-M implementation.

It seems that you are using the `thumbv7em-none-eabi[hf]` target. For blue-pill you must use the `thumbv7m-none-eabi` target. If you *are* using the `thumbv7m-none-eabi` target this could be a codegen bug in the compiler (in LLVM).

nallar, 2018-01-07:
I was using the wrong target.

Might be worth updating the documentation section "Optionally, set a default build target" [https://docs.rs/cortex-m-quickstart/0.1.8/cortex_m_quickstart/](cortex-m-quickstart) to mention that eabihf isn't always right depending on the target processor. From https://japaric.github.io/blue-pill/blue_pill/#usage you're directed to the quickstart crate without being warned to change it.

After changing the target to "thumbv7em-none-eabi" it still generates the uxtab instruction at opt-level = 0.

I rm -rf ~/.xargo and ran xargo clean. Is there anything else that might need removed?

```
Program received signal SIGTRAP, Trace/breakpoint trap.
cortex_m_rt::default_handler::h09ef7fbcd51f80a2 (ef=0x20004e00)
    at C:\Users\Luna\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-rt-0.3.8\src\lib.rs:459
459         asm::bkpt();
(gdb) bt
#0  cortex_m_rt::default_handler::h09ef7fbcd51f80a2 (ef=0x20004e00)
    at C:\Users\Luna\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-rt-0.3.8\src\lib.rs:459
#1  <signal handler called>
#2  0x08000252 in blue_pill_usb_keyboard::main::_$u7b$$u7b$closure$u7d$$u7d$::hbb85efdde22dada6 (_t=0x20004f87) at src\main.rs:109
#3  0x08005622 in cortex_m_rtfm::atomic::hc58ac6892de4c8fc (t=0x20004f97,
    f=...)
    at C:\Users\Luna\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-rtfm-0.2.2\src/lib.rs:110
#4  0x080001e2 in blue_pill_usb_keyboard::main::h89da4d3b38bebe0d ()
    at src\main.rs:102
#5  0x080055d4 in cortex_m_rt::lang_items::start::h934682ce0b6544a0 (
    main=0x80001bb <blue_pill_usb_keyboard::main::h89da4d3b38bebe0d>,
    _argc=0, _argv=0x0)
    at C:\Users\Luna\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-rt-0.3.8\src/lang_items.rs:31
#6  0x0800127e in main ()
(gdb) x/i 0x08000252
   0x8000252 <blue_pill_usb_keyboard::main::_$u7b$$u7b$closure$u7d$$u7d$::hbb85efdde22dada6+96>:        uxtab   r0, r0, r1
```

jonas-schievink, 2018-01-07:
The problem isn't "eabihf" but "thumbv7**e**m", you have to remove the e

nallar, 2018-01-07:
Thanks @japaric / @jonas-schievink  , it's working now :)

closed 2018-01-07 by nallar