rtic-rs/rtic #67: Language item required, but not found: panic_fmt

clebi, 2018-01-17:
Hello,

Trying to update to the last version of this crate, I had the following error at compilation : 

```
[clement@clement-desktop-home i2c_magnetic]$ xargo build 
   Compiling i2c_magnetic v0.1.0 (file:///home/clement/atom-projects/cortex/i2c_magnetic)
error: language item required, but not found: `panic_fmt`

error: aborting due to previous error

error: Could not compile `i2c_magnetic`.
```

I also updated the f3 crate to the 0.5.1 version.

japaric, 2018-01-17:
That sounds like the "abort-on-panic" feature has not been enabled for the cortex-m-rt crate. You can either enable that or provide your own `panic_fmt` lang item.

japaric, 2018-01-17:
> I also updated the f3 crate to the 0.5.1 version.

If you are depending on the f3 crate, instead of re-using the f3 Cargo project, then you'll need to enable the cortex-m-rt feature like this:

``` toml
[dependencies.cortex-m-rt]
features = ["abort-on-panic"]
version = "0.3.12"
```

clebi, 2018-01-17:
I have the following in my Cargo.toml : 

```
[dependencies.cortex-m-rt]
features = ["abort-on-panic"]
version = "0.3.0"
```

Here is my complete Cargo file : 

```
[package]
name = "i2c_magnetic"
version = "0.1.0"
authors = ["clement"]

[dependencies]
cortex-m = "0.4.0"
cortex-m-rtfm = "0.3.1"
f3 = "0.5.1"

[dependencies.cortex-m-rt]
features = ["abort-on-panic"]
version = "0.3.0"

[dependencies.cortex-m-semihosting]
version = "0.2.0"

[profile.release]
debug = true
lto = true

[profile.dev]
codegen-units = 1
incremental = false
```

japaric, 2018-01-17:
Give me a few minutes. Something definitively seems to be wrong in the f3 crate (it's missing the `extern crate cortex_m_rt`) but for some reason my example are building / linking without errors. 😕

japaric, 2018-01-17:
Just to rule out some rustc weirdness. What nightly version are you using? @clebi

japaric, 2018-01-17:
> ~~Something definitively seems to be wrong in the f3 crate~~

Actually the f3 crate is OK. f3 is pulling cortex-m-rt into mix through its stm32f30x dependency. Perhaps you really are missing the `extern crate f3` in your file? Do you have a branch for me to test?

clebi, 2018-01-17:
I am on the nightly rustc 1.25.0-nightly (da569fa9d 2018-01-16).

I have the extern crate f3 in my main.rs

```
#![feature(proc_macro)]
#![no_std]

extern crate cortex_m;
extern crate cortex_m_rtfm as rtfm;
extern crate cortex_m_semihosting as semihosting;
extern crate f3;
```

japaric, 2018-01-17:
I don't see any obvious problem and I can't repro locally with the f3 examples. The only thing that occurs to me if you got a "lemon" version for the cortex-m-rt crate. v0.3.10 and v0.3.11 are broken and already yanked but if you updated your lockfile before I got to yank them you may be using the broken version. I suggest forcing the cortex-m-rt version to be at least v0.3.12 in your Cargo.toml.

clebi, 2018-01-17:
My cortex-m-rt seems to be at version 0.3.12.

```
[[package]]
name = "cortex-m-rt"
version = "0.3.12"
source = "registry+https://github.com/rust-lang/crates.io-index"
dependencies = [
 "chrono 0.4.0 (registry+https://github.com/rust-lang/crates.io-index)",
 "cortex-m 0.3.1 (registry+https://github.com/rust-lang/crates.io-index)",
 "r0 0.2.2 (registry+https://github.com/rust-lang/crates.io-index)",
 "rustc_version 0.2.1 (registry+https://github.com/rust-lang/crates.io-index)",
]
```

I can send you an archive of the project.

japaric, 2018-01-17:
> I can send you an archive of the project.

Please do.

clebi, 2018-01-17:
Here it is: [i2c_magnetic.zip](https://github.com/japaric/cortex-m-rtfm/files/1640683/i2c_magnetic.zip)

japaric, 2018-01-17:
Oh, I see now. The "rt" feature of the stm32f30x needs to be enable or the cortex-m-rt won't be included in the dependency graph. You can work around the issue by adding these lines to your Cargo.toml:

``` toml
[dependencies.stm32f30x]
version = "0.6.0"
features = ["rt"]
```

But I think it would make most sense to re-export the "rt" feature through the stm32f30x-hal and f3 crates.

clebi, 2018-01-17:
It's working now ! thanks.

japaric, 2018-01-17:
> It's working now !

Yay!

> thanks

Thank *you* for reporting the issue and pointing holes in the crates.

I'm closing this issue though because this should be fixed in stm32f30x-hal and f3.

closed 2018-01-17 by japaric