rtic-rs/rtic #69: port the app! macro to the latest syn crate

japaric, 2018-02-12:
> port the app! macro to the latest syn crate

The latest version of the `syn` crate makes it easier to create parsers thanks to its parser
combinator (?) macros. It can also give better error (and warning) messages that have proper span
information. Let's port the `app!` macro to it to provide a better user experience.

Here's what needs to be done:

- Pretty much rewrite the `app!` parser in the [`rtfm-syntax`] crate. You'll have to replace
  `rtfm_syntax::App` and friends with a new struct that implements the [`Synom`] trait. You can find
  the specification of the `app!` macro syntax in the [crate documentation] of the
  `cortex-m-rtfm-macros` crate.

[crate documentation]: https://github.com/japaric/cortex-m-rtfm/blob/f76bdc285d70e1fe674843a6e329ae7707fdaa49/macros/src/lib.rs#L22-L169

[`rtfm-syntax`]: https://github.com/japaric/rtfm-syntax
[`Synom`]: https://docs.rs/syn/0.12.12/syn/synom/trait.Synom.html

- Update the code that converts the AST into a syntax checked AST. Basically update the
  `rtfm_syntax::check::app` function. You should try to keep the structs in `rtfm_syntax::check`
  unchanged -- unless there's some improvement that requires changing them.

- Update `cortex-m-rtfm-macros` to use the new version of `rtfm-syntax`. The changes should be
  minimal.

To get familiar with the parse combinator macros check out:

- The `syn` crate [documentation].
- The [`lazy_static!`] example
- A [`mat!`] macro I recently wrote.

[documentation]: https://docs.rs/syn/0.12.12/syn/#parsing-a-custom-syntax-using-combinators
[`lazy_static!`]: https://github.com/dtolnay/syn/tree/master/examples/lazy-static
[`mat!`]: https://github.com/japaric/mat/tree/3dd15142891eb5da256722a23f334b6845d822fb/macros

Be sure to enable the following features in your dependencies:

``` toml
[dependencies.syn]
version = "0.12.12"
features = ["full"]

[dependencies.proc-macro2]
features = ["nightly"]
version = "0.2.2"
```

If you want to give this a stab leave a comment. If no one volunteers I'll probably get to this
before the week ends.

creativcoder, 2018-02-16:
Hey, @japaric I would like to help on this one. I am first trying to run one of the examples (one-task) in `cortex-m-rtfm` crate which points to my local copy of `rtfm-syntax`, just to get started with things.
On running `cargo run --example one-task`, it fails and
`compiletest_rs` does not compile for me: 
```
➜  cortex-m-rtfm git:(master) ✗ cargo run --example one-task
   Compiling compiletest_rs v0.3.6
error[E0433]: failed to resolve. Could not find `OutputFormat` in `test`
  --> /home/creativcoder/.cargo/registry/src/github.com-1ecc6299db9ec823/compiletest_rs-0.3.6/src/lib.rs:97:41
   |
97 |         format: if config.quiet { test::OutputFormat::Terse } else { test::OutputFormat::Pretty },
   |                                         ^^^^^^^^^^^^ Could not find `OutputFormat` in `test`

error[E0433]: failed to resolve. Could not find `OutputFormat` in `test`
  --> /home/creativcoder/.cargo/registry/src/github.com-1ecc6299db9ec823/compiletest_rs-0.3.6/src/lib.rs:97:76
   |
97 |         format: if config.quiet { test::OutputFormat::Terse } else { test::OutputFormat::Pretty },
   |                                                                            ^^^^^^^^^^^^ Could not find `OutputFormat` in `test`

error[E0560]: struct `test::TestOpts` has no field named `format`
  --> /home/creativcoder/.cargo/registry/src/github.com-1ecc6299db9ec823/compiletest_rs-0.3.6/src/lib.rs:97:9
   |
97 |         format: if config.quiet { test::OutputFormat::Terse } else { test::OutputFormat::Pretty },
   |         ^^^^^^^ `test::TestOpts` does not have this field
   |
   = note: available fields are: `list`, `filter`, `filter_exact`, `run_ignored`, `run_tests` ... and 8 others

error: aborting due to 3 previous errors

error: Could not compile `compiletest_rs`.
```
I am on rust `rustc 1.25.0-nightly (15a1e2844 2018-01-20)`
Which nightly version should i use ?

japaric, 2018-02-16:
> I would like to help on this one.

Awesome.

compiletest_rs v0.3.6 (call `cargo update` to update the lockfile) and nightly-2018-02-11 works for me. But don't run the examples like that because they are not mean to be run on x86_64. You can run the test suite by doing exactly what Travis does:

``` console
$ # run compile-fail tests on the hosts
$ TARGET=x86_64-unknown-linux-gnu bash ci/script.sh

$ # check that the examples still compile
$ TARGET=thumbv7m-none-eabi bash ci/script.sh
```

creativcoder, 2018-02-16:
Cool, tests are green now. Off to porting :)

commit 7fdf16ea (2018-04-16): update parser

closes #69

this doesn't change functionality per se but improves diagnostics in some cases. Some hard errors
have becomes warnings, for example: when `resources` is empty, or when `idle.path` is set to the
default `idle` path.
commit 6909f5f7 (2018-04-16): Merge #71

71: update parser r=japaric a=japaric

closes #69

this doesn't change functionality per se but improves diagnostics in some cases. Some hard errors
have becomes warnings, for example: when `resources` is empty, or when `idle.path` is set to the
default `idle` path.
commit 5ff9076e (2018-04-16): Merge #71

71: update parser r=japaric a=japaric

closes #69

this doesn't change functionality per se but improves diagnostics in some cases. Some hard errors
have becomes warnings, for example: when `resources` is empty, or when `idle.path` is set to the
default `idle` path.
PR #71 (2018-04-16): update parser
closes #69

this doesn't change functionality per se but improves diagnostics in some cases. Some hard errors
have becomes warnings, for example: when `resources` is empty, or when `idle.path` is set to the
default `idle` path.
closed 2018-04-16 by bors[bot]