rtic-rs/rtic #73: Documentation for version 0.3.2 does not exist

CuriouslyCurious, 2018-04-23:
The documentation [link](https://docs.rs/cortex-m-rtfm/0.3.2/cortex_m_rtfm/) for version 0.3.2 on [https://crates.io/crates/cortex-m-rtfm](https://crates.io/crates/cortex-m-rtfm) does not work. It simply returns a page with: 
```
The requested resource does not exist
```
By changing the `0.3.2` in the link to for example `0.3.1` you get the actual documentation: [https://docs.rs/cortex-m-rtfm/0.3.1/cortex_m_rtfm/](https://docs.rs/cortex-m-rtfm/0.3.1/cortex_m_rtfm/).

wez, 2018-05-12:
The docs are failing to build with errors like this:

```
error[E0433]: failed to resolve. Could not find `Group` in `proc_macro`
  --> .cargo/registry/src/github.com-1ecc6299db9ec823/proc-macro2-0.3.6/src/unstable.rs:63:45
   |
63 |                 let mut group = proc_macro::Group::new(delim, tt.stream.inner.0);
   |                                             ^^^^^ Could not find `Group` in `proc_macro`

```

edarc, 2018-08-02:
In fact, this crate (or rather one of its dependencies) is not compiling at all for me on `rustc 1.28.0-nightly (84804c387 2018-06-26)` and I guess the broken documentation is a symptom.

I am gathering from `proc-macro2`'s [changelog](https://github.com/alexcrichton/proc-macro2/commit/f388843b8a92b5523c30d51f1da436a05ebb5551), this is because rustc's proc macro API changed and so `proc-macro2 = "0.3.x"` are all unusable on recent nightlies. I haven't dug deeply yet but the names of the altered APIs referenced in that commit message appear in the compilation errors. Assuming that's the cause, I suppose the fix would be to migrate `cortex-m-rtfm-macros` to `proc-macro2 = "0.4"`.

Is anyone working on that or can I take a crack at it? (Or maybe I'm missing something)

japaric, 2018-11-03:
The [book] and the [API reference] are now hosted on GH pages.

[book]: https://japaric.github.io/cortex-m-rtfm/book/
[API reference]: https://japaric.github.io/cortex-m-rtfm/api/rtfm/index.html

closed 2018-11-03 by japaric