rtic-rs/rtic #76: [book] does it worth to pack tasks into crates?

puzrin, 2018-05-08:
Upcoming book contains very good messaging example: https://japaric.github.io/cortex-m-rtfm/user/messages.html. There are some common cases (like block transfers via uart/serial/i2c buses), had to be repeated in many project. It looks attractive to arrange those to crates somehow. Questions are:

- Are there any recommendations how to arrange such RTFM extensions?
- Should we move in such direction or isn't it better to push such feature into HAL (or some HAL extender)?
  - If yes, what should we know about intersecting such libs (with interrupts and processing inside) with RFTM priorities logic?

In other words, as user, i'd like to have "ready solution to read strings from UART" (that's only an example). And use it in RTFM. What is the optimal way to develop such crates? May be worth add some words about into the book.

PS. i'm not sure. Feel free to close anytime.

perlindgren, 2018-05-08:
Hi

Well as of now, the app! is a centralized system description. This allows us to do the static analysis and code generation based on system wide information.

To embed interrupt bindings into modules or even crates requires gathering information from different modules/crates.

As of today, that is not possible using procedural macros. It is possible to do as a compiler plugin (like the linters), but those are invoked at a later stage in the rustc compiler and not intended to generate code.

Ongoing work on the Rust compiler might in the future allow what you are after. (We did some initial experiments of using the build script to parse syntax extensions (to gather crate wide info but not cross crate info), but that I believe not to be the best sustainable solution.)

So essentially, for now we are blocked by limitations of the Rust lang/and rustc implementation to a centralized system description. This does not prevent you from modularizing functionality, but interrupt bindings cannot (currently) be defined outside the app!.

Best

  Per

bergus, 2018-05-24:
I wouldn't like RTFM to gather information from all files in the crate (or whole dependency tree). That makes it hard to find all definitions - and conflicts would lead to difficult-to-reproduce problems.

But I did imagine some kind of plugin system where a crate could define a macro that would then be explicitly passed to `app!`, which would expand (call) the macro and incorporate the resulting declarations. Is that a feasible approach?

perlindgren, 2018-05-24:
Hi

Having ort of “local” app macros could perhaps be done in the future. If I got it right there is a proposal regarding a new test framework that allows to “export” and “gather” items. For what I understand its only within a crate (not between crates).

In the experimental “reactive objects” approach we used external files (and a build script) to emulate this behaviour. (Also not cross crate atm.)

Maybe Jorge sits on more info on the progress of the new test framework and how we could best adopt this.

Best,
  Per

et1975, 2018-06-25:
I'm totally new to RTFM, but I'd think that plugin system would be a step backwards in terms of compile-time verifiability. A bit of explicit assembly at the top in exchange for increased guarantees seems preferable.

korken89, 2019-12-29:
If this conversation is not done, please reopen. Closing [issue cleanup]

closed 2019-12-29 by korken89