rtic-rs/rtic #81: cannot set high priority for scheduled tasks

burrbull, 2018-06-22:
I use RTFM 0.4 from
https://github.com/japaric/stm32f103xx-hal/tree/rtfm-up

But when I try set `priority` different from 1 for task that not associated with interrupt, program compiling fails with error:
```
error: proc macro panicked
   --> src/main.rs:344:1
    |
344 | / app! {
345 | |     device: stm32f103xx,
346 | |
347 | |     resources: {
...   |
395 | |     },
396 | | }
    | |_^
    |
    = help: message: not enough free interrupts
```

bergus, 2018-06-23:
You cannot have a task that is not associated with an interrupt. However, RTFM can do the assignment for you (and also put multiple tasks on the same interrupt if they have the same priority); you will need to provide a list like `free_interrupts: [EXTI0],` for that though.

burrbull, 2018-06-23:
Any interrupt that I will not use, or some particular?
I think this should be added to user manual in book.
Thanks.

japaric, 2018-11-03:
This is now covered in [the book](https://japaric.github.io/cortex-m-rtfm/book/by-example/tasks.html).

closed 2018-11-03 by japaric