rtic-rs/rtic #84: How to claim resources without a token?

bergus, 2018-07-10:
I am trying to use the [clean architecture](https://8thlight.com/blog/uncle-bob/2012/08/13/the-clean-architecture.html) for my device - i.e. the main application logic is not dependent on the hardware, but is written with a generic output trait that can be instantiated with different types. 
```rust
pub trait Output {
    fn write(Message);
}
pub struct App<O: Output> {
    out: O,
    …
}
impl<O: Output> App<O> {
    pub fn new(out: O) -> Self {
        Self { out, … }
    }
    pub fn process(&mut self, msg: Message) {
        … // heavy lifting
        self.out.write(…);
        …
    }
}
```
For testing, I can instantiate the `Output` with a growing vector where I can inspect the results after processing various example inputs.
In the application, I can instantiate the `Output` with a producer that writes into a buffer, getting consumed from the serial port interrupt:
```rust
// pseudo code :-)
use my_library::App;
use buffer::{Buffer, BufferConsumer, BufferProducer};
use hal::Uart;

struct SerialApp {
    input: BufferConsumer,
    app: App<SerialBufferProducer>
}
struct SerialBufferProducer(BufferProducer);
rtfm::app! {
    resources: {
        static app: SerialApp;
        static serial_rx: BufferProducer;
        static serial_tx: BufferConsumer;
        static uart: Uart;
    },
    idle: {
        resources: [app],
    },
    tasks: {
        USART1: {
            path: rxtx,
            resources: [uart, serial_rx, serial_tx],
            enabled: true,
        }
    },
}
fn init(mut p: init::Peripherals) -> init::LateResources {
    static mut rx_buffer: Buffer = Buffer::empty();
    static mut tx_buffer: Buffer = Buffer::empty();
    let (rx_consumer, rx_producer) = unsafe { rx_buffer.split() };
    let (tx_consumer, tx_producer) = unsafe { tx_buffer.split() };
    init::LateResources {
        app: SerialApp::new(rx_consumer, SerialBufferProducer(tx_producer)),
        serial_rx: rx_producer,
        serial_tx: tx_consumer,
        uart: Uart::init(p.USART1),
    }
}
fn idle(t: &mut Threshold, mut r: idle::Resources) -> ! {
    loop {
        if let Some(msg) = r.serial_app.input.read() {
            r.serial_app.app.process(msg);
        }
    }
}
pub fn rxtx(_: &mut ::Threshold, mut r: ::USART1::Resources) {
    if let Some(byte) = r.uart.received() {
        r.serial_rx.write(byte);
    }
    if r.uart.can_send() {
        if let Some(byte) = r.serial_tx.read() {
            r.uart.send(byte);
        } else {
            r.uart.disable_tx_event();
        }
    }
}
```
Here's the catch: the `SerialBufferProducer` wrapper that implements the app `Output` is supposed to enable the uart tx event after writing to the buffer, and it would need to claim the `uart` resource for that!

Let's assume that
* I want to use the `app` in other tasks as well, so I can't just instantiate it within `idle` and use its token and uart resource in the construction of the `SerialBufferProducer`.
* there is no other way to guarantee atomical access, like splitting the `uart` resource or using bit banding - I really need to prevent other `uart`-using interrupts

I think I need a way to access a `uart` resource in the `init` function so that I can put it into the `SerialBufferProducer`, and then a way to claim it without being passed a token. The clean architecture prevents me from passing the token that is accessible at the call site of `process` through to the `write` call. What do you think?

japaric, 2018-11-03:
> How to claim resources without a token?

That's how `Mutex.lock` works in v0.4.0; you don't need a token -- the `Threshold` token is gone in v0.4.0.

> I think I need a way to access a uart resource in the init function

This is not possible. Resource proxies (the `struct`s that implement the `Mutex` trait) are tied to the stack frame of the task that owns them; they are not `Send`, either, so even if you could create one in `init` you would not be able to move into any task. 

If you need fine-grained locks, i.e. a struct that contains a resource proxy, you'll have to write a generic struct (e.g. `struct Foo<R> where R: Mutex<Data = u32> { resource: R, state: State }` and re-instantiate it in every task where you need it.

bergus, 2018-11-06:
Thanks, I'll take a look at mutexes. It'll be hard to do initialisation different, though.

japaric, 2019-05-08:
I think the original question has been answer. If you have additional questions please open a new issue.

closed 2019-05-08 by japaric