rtic-rs/rtic #86: Update to cortex-m 0.5

djmcgill, 2018-08-28:
With this crate on `cortex-m-0.4`, it's impossible to use any tooling which also includes `cortex-m-0.5` - you get errors like
```
warning: Linking globals named 'CORE_PERIPHERALS': symbol multiply defined!
error: failed to load bc of "cortex_m.22u1h9r5-cgu.2":
```
Because (I'm pretty sure) both versions of the cortex-m libraries define those symbols and they clash. An example of this in action is:
https://travis-ci.org/wez/atsamd21-rs/builds/421535430
```
  = note: rust-lld: error: duplicate symbol: CORE_PERIPHERALS
          >>> defined at mod.rs:148 (/home/travis/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.4.3/src/peripheral/mod.rs:148)
          >>>            cortex_m-89a330ceffd88411.cortex_m.6bne0cac-cgu.0.rcgu.o:(CORE_PERIPHERALS) in archive /home/travis/build/wez/atsamd21-rs/metro_m0/target/thumbv6m-none-eabi/debug/deps/libcortex_m-89a330ceffd88411.rlib
          >>> defined at mod.rs:154 (/home/travis/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.5.6/src/peripheral/mod.rs:154)
          >>>            cortex_m-202b80cd0a33d787.cortex_m.er0kkjw4-cgu.0.rcgu.o:(.bss.CORE_PERIPHERALS+0x0) in archive /home/travis/build/wez/atsamd21-rs/metro_m0/target/thumbv6m-none-eabi/debug/deps/libcortex_m-202b80cd0a33d787.rlib         
```

ykomatsu, 2018-09-01:
I made a modified cortex-m-rtfm that depends on cortex-m 0.5.6:

https://github.com/ykomatsu/cortex-m-rtfm/tree/update

japaric, 2018-11-03:
Done in #97

closed 2018-11-03 by japaric