rtic-rs/rtic #89: Which nightly should I be using?

daniel-dsouza, 2018-09-21:
I cannot get anything to link with `rustc 1.30.0-nightly (3bc2ca7e4 2018-09-20)`. More specifically the GNU linker fails with:

```  
= note: arm-none-eabi-ld:
          BUG(cortex-m-rt): the reset vector is missing
          arm-none-eabi-ld:
          BUG(cortex-m-rt): the exception vectors are missing
```

and the rustc lld cannot  even find lld, even with `llvm-tools-preview-x86_64-unknown-linux-gnu`

I would really like try Rust for a project, but it might be better to use C++ instead if I cannot even get my environment setup.

TeXitoi, 2018-09-21:
Can you point to a complete project? You are lacking the runtime that set these vectors.

daniel-dsouza, 2018-09-30:
I was trying to follow [this ](https://github.com/lupyuen/stm32-blue-pill-rust) tutorial, on Ubuntu 18.04

TeXitoi, 2018-09-30:
https://github.com/japaric/cortex-m-rtfm/issues/90 might be relevant.

TeXitoi, 2018-09-30:
https://github.com/TeXitoi/blue-pill-quickstart can also interest you.

daniel-dsouza, 2018-09-30:
Thanks, I'll give a shot when I have access to the hardware.

japaric, 2018-11-03:
v0.4.x compiles with 1.31-beta. Let us know in a new issue if you have problems with that version. v0.3.x is EOL and no longer supported.

closed 2018-11-03 by japaric