rtic-rs/rtic #90: minimal example fails to build

timbod7, 2018-09-24:
I've copied

https://github.com/japaric/cortex-m-rtfm/blob/master/examples/zero-tasks.rs

 into my own repo, and adapted it to use the stm32f4 crate. The resulting code is here: 

https://github.com/timbod7/rust-stm32f4-examples/blob/master/minimal-rtfm/src/main.rs

this fails to build with a suprising (to me) error:

```

timbod7, 2018-09-24:
Not sure I understand what you mean by "use #87".

I've simplified the dependendecies, and use cortex-m-rt 0.3:

```
[dependencies.stm32f4]
version = "0.2.3"
features = ["stm32f407", "rt"]

[dependencies.cortex-m]
version = "0.5.7"

[dependencies.cortex-m-rt]
version = "0.3"

[dependencies]
cortex-m-rtfm = "0.3.4"

[dependencies.panic-abort]
version = "0.3.0"

TeXitoi, 2018-09-25:
stm32f4 v0.2.3 depends on cortex-m v0.5.

I mean you can use the branch of the pull request:
```
cortex-m-rtfm = { git = "https://github.com/ykomatsu/cortex-m-rtfm", branch = "update" }
```

timbod7, 2018-09-25:
Thanks so much! That approach did eventually work.

closed 2018-09-25 by timbod7

TeXitoi, 2018-09-25:
Welcome.