rtic-rs/rtic #93: Proper solution for the use of  embedded-nrf24l01 driver in RTFM application

geomatsi, 2018-10-23:
Hi,

I have been experimenting with a simple RTFM application using [embedded-nrf24l01](https://github.com/astro/embedded-nrf24l01) crate. That crate implements driver for nRF24L01 radio chip using  some kind of 'mutate object using self move' semantics:
```
// create driver in 'standby' state and perform initial configuration
let mut s = NRF24L01::new(ce, cs, spi).unwrap();
s.set_frequency(10).unwrap();
...
// 'move' into Tx state to send packets
let t = s.tx().unwrap()
t.send(&[0x1, 0x2]).unwrap();
// 'move' back into standby state after transmission is acknowledged
let s = t.standby().unwrap()
```
I had some difficulties with keeping this kind of driver as a resource in RTFM application. So far I could make it work only using `RefCell<Option<...>>` aproach:
```
...

type Standby = ...;

app! {
    device: hal::stm32f103xx,

    resources: {
        static NRF: RefCell<Option<Standby>>;
        ...
    },
...
fn timer_handler(_t: &mut Threshold, mut r: TIM3::Resources) {
    let data: [u8; 10] = [0x31; 10];
   
    let s = match r.NRF.replace(None) {
        None => panic!("booo"),
        Some(s) => {
            let mut t = s.tx().unwrap();
            t.send(&data).unwrap();
            t.standby().unwrap()
        }
    };

    r.NRF.replace(Some(s));
}
```
Full source is available [here](https://github.com/geomatsi/rust-blue-pill-tests/blob/master/src/bin/nrf24-rtfm-test1.rs). Is there another, more appropriate and elegant way to handle such a driver in RTFM applications ?

Regards,
Sergey

japaric, 2018-11-03:
> embedded-nrf24l01 driver

Type state for the IO is an interesting choice.

You have to do the option dance, which is what you are already doing. You don't need the `RefCell` though; you should be able to use `Option.take`. Here's an `Option.take` version with no unwraps / panics:

``` rust
fn foo(driver: &mut Option<Driver<StandBy>>) {
    // start dance
    if let Some(standby) = driver.take() {
        // change state
        let read = standby.read();

        // do stuff with `read`

        // change state
        let standby = read.standby();

        // end dance
        *driver = Some(standby);
    }
}

use std::marker::PhantomData;

struct Driver<STATE> {
    _state: PhantomData<STATE>,
}

impl<S> Driver<S> {
    fn read(self) -> Driver<Read> {
        Driver {
            _state: PhantomData,
        }
    }

    fn standby(self) -> Driver<StandBy> {
        Driver {
            _state: PhantomData,
        }
    }
}

struct StandBy;
struct Read;
```

geomatsi, 2018-11-04:
Thanks! This approach works fine. Indeed, the use of RefCell was redundant here, Option.take does the same thing. Closing the issue...

closed 2018-11-04 by geomatsi