rtic-rs/rtic #95: Build issues with STM32F042 target

osterwood, 2018-10-31:
Hi,

I'm new to rust and the embedded rust eco-system, so forgive me if I've missed a simple initial step which would resolve my issues.  I've uploaded a single task RTFM application here:
    https://github.com/osterwood/rtfm-app
It should blink an LED on a STM32F042 board I have.  It's based on the `rtfm-one.rs`example in this repository.

I've run into a compile issue that I don't know how to resolve.
```                                                    
error[E0432]: unresolved import `_initResources`                                                                                                             
  --> examples/rtfm-one.rs:28:1                                                                                                                              
   |                                                                                                                                                         
28 | / app! {                                                                                                                                                
29 | |     device: stm32f0::stm32f0x2,                                                                                                                       
30 | |                                                                                                                                                       
31 | |     // Here data resources are declared                                                                                                               
...  |                                                                                                                                                       
57 | |     }                                                                                                                                                 
58 | | }                                                                                                                                                     
   | |_^ no `_initResources` external crate                                                                                                                  
                                                                                                                                                             
error[E0433]: failed to resolve. Could not find `init` in `{{root}}`                                                                                         
  --> examples/rtfm-one.rs:28:1                                                                                                                              
   |                                                                                                                                                         
28 | / app! {                                                                                                                                                
29 | |     device: stm32f0::stm32f0x2,                                                                                                                       
30 | |                                                                                                                                                       
31 | |     // Here data resources are declared                                                                                                               
...  |                                                                                                                                                       
57 | |     }                                                                                                                                                 
58 | | }                                                                                                                                                     
   | |_^ Could not find `init` in `{{root}}`                                                                                                                 
                                                                                                                                                             
error[E0425]: cannot find value `ON` in the crate root                                                                                                       
  --> examples/rtfm-one.rs:28:1                                                                                                                              
   |                                                                                                                                                         
28 | / app! {                                                                                                                                                
29 | |     device: stm32f0::stm32f0x2,                                                                                                                       
30 | |                                                                                                                                                       
31 | |     // Here data resources are declared                                                                                                               
...  |                                                                                                                                                       
57 | |     }                                                                                                                                                 
58 | | }                                                                                                                                                     
   | |_^ not found in the crate root                                                                                                                         
                                                                                                                                                             
error: unused import: `stm32f0::stm32f0x2`                                                                                                                   
  --> examples/rtfm-one.rs:25:5                                                                                                                              
   |                                                                                                                                                         
25 | use stm32f0::stm32f0x2;                                                                                                                                 
   |     ^^^^^^^^^^^^^^^^^^                                                                                                                                  
   |                                                                                                                                                         
note: lint level defined here                                                                                                                                
  --> examples/rtfm-one.rs:3:9                                                                                                                               
   |                                                                                                                                                         
3  | #![deny(warnings)]                                                                                                                                      
   |         ^^^^^^^^                                                                                                                                        
   = note: #[deny(unused_imports)] implied by #[deny(warnings)]                                                                                              
                                                                                                                                                             
error[E0308]: mismatched types                                                                                                                               
[... cut ...]
                                                                                                                                                             
error: aborting due to 5 previous errors                                                                                                                                                          
```

It seems there are multiple issues with the `app!` macro.  Most of the examples I've found for RTFM are based on the stm32f103xx and stm32f103_hal crates, not stm32f0.  Is there something different about the stm32f0 crate which breaks RTFM?  

Thanks in advance for any ideas you have.

TeXitoi, 2018-10-31:
Maybe just trying to use edition 2015, i.e. remove this line in Cargo.toml:
```rust
edition = "2018"
```

osterwood, 2018-10-31:
That's better!  Now down to two errors (once I changed `device stm32f0::stm32f0x2` to `device stm32f0x2` and made the end of init a () return (instead of returning the `init::LateResources` struct).

```
error[E0308]: mismatched types                                                                                                                               
  --> examples/rtfm-one.rs:28:1                                                                                                                              
   |                                                                                                                                                         
28 | / app! {                                                                                                                                                
29 | |     device: stm32f0x2,                                                                                                                                
30 | |                                                                                                                                                       
31 | |     // Here data resources are declared                                                                                                               
...  |                                                                                                                                                       
57 | |     }                                                                                                                                                 
58 | | }                                                                                                                                                     
   | |_^ expected struct `_initLateResources`, found ()                                                                                                      
   |                                                                                                                                                         
   = note: expected type `for<'r> fn(init::Peripherals, _initResources<'r>) -> _initLateResources`                                                           
              found type `for<'r> fn(init::Peripherals, _initResources<'r>) {init}`                                                                          
                                                                                                                                                             
error[E0308]: mismatched types                                                                                                                               
  --> examples/rtfm-one.rs:28:1                                                                                                                              
   |                                                                                                                                                         
28 | / app! {                                                                                                                                                
29 | |     device: stm32f0x2,                                                                                                                                
30 | |                                                                                                                                                       
31 | |     // Here data resources are declared                                                                                                               
...  |                                                                                                                                                       
57 | |     }                                                                                                                                                 
58 | | }                                                                                                                                                     
   | |_^ expected u32, found u8                                                                                                                              
help: you can cast an `u8` to `u32`, which will zero-extend the source value                                                                                 
   |                                                                                                                                                         
28 | app! {                                                                                                                                                  
29 |     device: stm32f0x2,                                                                                                                                  
30 |                                                                                                                                                         
31 |     // Here data resources are declared                                                                                                                 
32 |     //                                                                                                                                                  
33 |     // Data resources are static variables that are safe to share across tasks                                                                          
 ...                                                                                                                                                                                                                                                                                                                
error: aborting due to 2 previous errors   
```

TeXitoi, 2018-10-31:
init must return LateResources, that's tthe first error.

TeXitoi, 2018-10-31:
You'll aalso need the update branch of RTFM as commented in your Cargo.toml to use cortex-m 0.5

osterwood, 2018-10-31:
Thank so much for the help.  I had pulled the init signature from an example which didn't use LateResources and neglected to change it when I added one.

I think I'm nearly there.  Now have this error about SYS_TICK
```
error[E0599]: no variant named `SYS_TICK` found for type `hal::stm32::Interrupt` in the current scope                                                        
  --> examples/rtfm-one.rs:27:1                                                                                                                              
   |                                                                                                                                                         
27 | app! {                                                                                                                                                  
   | ^ variant not found in `hal::stm32::Interrupt`                                                                                                          
                                                                                                                                                             
error: aborting due to previous error              
```

I've looked through rtfm, stm32-rs, stm32f042-hal, stm32f103xx -- and I don't see where SYS_TICK is defined.  stm32f103xx now only uses them in outdated examples.  Should I revert to using a hardware timer?

geomatsi, 2018-10-31:
After changing this name it is possible to go a bit further. Next error looks like the following:
```
error[E0308]: mismatched types                                                                                                                                                                             
  --> examples/rtfm-one.rs:28:1                                                                                                                                                                            
   |                                                                                                                                                                                                       
28 | app! {                                                                                                                                                                                                
   | ^ expected u32, found u8                                                                                                                                                                              
help: you can cast an `u8` to `u32`, which will zero-extend the source value                                                                                                                               
   |                                                                                                                                                                                                       
28 | a.into()pp! {                                                                                                                                                                                         
   | ^^^^^^^^                                                                                                                                                                                              
                                                                                                                                                                                                           
error: aborting due to previous error                                                                                                                                                                      
                                                                                                                                                                                                           
For more information about this error, try `rustc --explain E0308`.
```
On the first glance it looks like cortex-m-rtfm type issue for task priority: task priority is u8 while scb.shpr[] is [u32]. The following fix on top of [PR#87](https://github.com/japaric/cortex-m-rtfm/pull/87) helps to go further:
```
diff --git a/macros/src/trans.rs b/macros/src/trans.rs
index 894abd7..08b8a90 100644
--- a/macros/src/trans.rs
+++ b/macros/src/trans.rs
@@ -366,7 +366,7 @@ fn init(app: &App, main: &mut Vec<TokenStream>, root: &mut Vec<TokenStream>) {
                 }
 
                 let nr = e.nr();
-                let priority = task.priority;
+                let priority = task.priority as u32;
                 exceptions.push(quote! {
                     let prio_bits = #device::NVIC_PRIO_BITS;
                     let hw = ((1 << prio_bits) - #priority) << (8 - prio_bits);
```
However something is still wrong:
```
error: index out of bounds: the len is 2 but the index is 11                                                                                                                                               
  --> examples/rtfm-one.rs:28:1                                                                                                                                                                            
   |                                                                                                                                                                                                       
28 | app! {                                                                                                                                                                                                
   | ^                                                                                                                                                                                                     
   |                                                                                                                                                                                                       
   = note: #[deny(const_err)] on by default                                                                                                                                                                
                                                                                                                                                                                                           
error: aborting due to previous error       
```
Build can be completed by adding `#![allow(const_err)]`, but this hack is definitely wrong since that check exists somewhere in cortex-m-rtfm for reason.

geomatsi, 2018-11-01:
Just for the record, all these issues pop up only when we try to use SysTick timer. I have several cortex-m-rtfm examples for stm32f030 and stm32f103 using TIM timers. Those examples work like a charm w/o any compilation issues with the same list of dependencies in Cargo.toml.

geomatsi, 2018-11-01:
Finally, the root cause is as follows: cortex-m-rtfm calculates index for writing to scb.shpr[] in a way which is suitable for armv7, but not for armv6. The difference is clear from file src/peripheral/scb.rs from cortex-m source tree: see function set_priority

Quick hack (specific to armv6 only) which makes the app work on stm32f0 is as follows:
```
diff --git a/macros/src/trans.rs b/macros/src/trans.rs
index 894abd7..1f91a24 100644
--- a/macros/src/trans.rs
+++ b/macros/src/trans.rs
@@ -366,11 +366,11 @@ fn init(app: &App, main: &mut Vec<TokenStream>, root: &mut Vec<TokenStream>) {
                 }
 
                 let nr = e.nr();
-                let priority = task.priority;
+                let priority = task.priority as u32;
                 exceptions.push(quote! {
                     let prio_bits = #device::NVIC_PRIO_BITS;
                     let hw = ((1 << prio_bits) - #priority) << (8 - prio_bits);
-                    scb.shpr[#nr - 4].write(hw);
+                    scb.shpr[(#nr - 8) / 4].write(hw);
                 });
             }
             Kind::Interrupt { enabled } => {
```

@japaric , does it makes sense to work on a proper fix ? or this is a known issue and fix is somewhere on its way to cortex-m-rtfm/master ?

japaric, 2018-11-03:
@osterwood I would suggest reading through [RTFM book](https://japaric.github.io/cortex-m-rtfm/book/) and running the QEMU examples for now. You'll need to wait until svd2rust v0.14.0 is released and the device crates in the ecosystem catch up before you can run rtfm v0.4.0 on hardware.

@geomatsi v0.4.0 is using [SCB.set_priority](https://docs.rs/cortex-m/0.5.8/cortex_m/peripheral/struct.SCB.html#method.set_priority), which should do the right thing on ARMv6-M.

osterwood, 2018-11-09:
@japaric and @geomatsi Thanks for your help on this.  Looks like v0.4.0 will be really nice to work with.  In the meantime, I think I'll move to an armv7 MCU for my prototyping & testing.

japaric, 2018-12-16:
I'm going to close this issue since it's about v0.3.x, which is no longer supported / maintained.

closed 2018-12-16 by japaric