rtic-rs/rtic #102: Translations for RTFM book

burrbull, 2018-11-06:
I've translated RTFM book to [Russian](https://github.com/burrbull/cortex-m-rtfm/tree/book-ru). Are you interesting in multilingual support of books, this one and others?

japaric, 2018-11-14:
Nice! I'd be happy to land the translation in this repo. I think mdbook has some multilingual support, but I have never used it though. Are you familiar with it?

> Are you interesting in multilingual support of books, this one and others?

Depends on what books you mean by others. If you mean the books in docs.rust-embedded.org I think it would best to open an issue in the rust-embedded/wg repo and cc the resources team.

burrbull, 2018-11-14:
No. I found [this issue](https://github.com/rust-lang-nursery/mdBook/issues/5) and couldn't understand how to add language to mdbook.

Yes, I mean Embedded Book.

japaric, 2019-02-08:
@burrbull sorry for taking so long to get back to you. I have created a skeleton for the Russian translation in the [translation] branch. If you are still up for this you can start from the "translation" branch and should mostly just need to drop in your translation into the `ru/src` directory. You can test the setup by running `bash ci/after-success.sh` -- that will render both books in a temporary directory.

[translation]: https://github.com/japaric/cortex-m-rtfm/tree/translation

burrbull, 2019-02-08:
I've made PR #138 .

Can you create similar skeleton for Embedded Book?

japaric, 2019-02-10:
@burrbull Thanks! I *can* create the skeleton but unlike here I can't make independent decisions for the embedded book. Please create an issue about the translation and cc the resources team. They have to give their approval before I can create the skeleton.

japaric, 2019-02-11:
Translation is up at https://japaric.github.io/cortex-m-rtfm/book/ru/

@burrbull thanks for translating this! @little-arhat thanks for reviewing the translation!

closed 2019-02-11 by japaric