rtic-rs/rtic #103: Macro expansion error information

MabezDev, 2018-11-09:
Hey, 

I am working on porting a project to rtfm v4, one thing I have noticed is that any errors (init and interrupt functions) the errors are reported at the app macro attribute, for example putting `let i: () = 0u32;` into `init` produced this rustc output:

```rust
error[E0308]: mismatched types
  --> src/main.rs:64:1
   |
64 | #[app(device = crate::stm32l4x2)]
   | ^ expected (), found u32
   |
   = note: expected type `()`
              found type `u32`
```

It not a huge issue, but it would be nice to know exactly where the errors are.

`rustc --version
rustc 1.31.0-nightly (a66dc8a14 2018-10-22)`

On a side note, any idea when svd2rust 0.14.x will be shipping?

japaric, 2018-11-14:
You are probably hitting [this compiler bug](https://github.com/rust-lang/rust/issues/43081#issuecomment-408800102). rustfmt-ed code usually works for me:

``` rust
#![no_main]
#![no_std]

extern crate panic_halt;

use rtfm::app;

#[app(device = lm3s6965)]
const APP: () = {
    #[init]
    fn init() {
        let x: () = 0;
        //          ^ expected (), found integral variable
    }
};
```

> On a side note, any idea when svd2rust 0.14.x will be shipping?

Soon (TM)

MabezDev, 2018-12-15:
Just coming back to this, in prep for stm32-rs 0.5.0 release that is built with svd2rust0.14.

I've managed to track it down, and I believe it's rustfmt causing the issue.

It takes this line:

```rust
static mut TOUCH: hal::tsc::Tsc<hal::gpio::gpiob::PB4<hal::gpio::Alternate<hal::gpio::AF9, hal::gpio::Output<hal::gpio::OpenDrain>>>> = ();
```

and adds commas as it restructures the line.

```rust
static mut TOUCH: hal::tsc::Tsc<
        hal::gpio::gpiob::PB4<
            hal::gpio::Alternate<hal::gpio::AF9, hal::gpio::Output<hal::gpio::OpenDrain>>,
        >,
    > = ();
```

This must fudge the macro information somewhere. Removing the commas is not enough to get it working again, it must be restored to the original one line format to get the correct error information.

As the code compiles, I assume rustfmt is allowed to do this, so does this need to be fixed in the macro parsing?

japaric, 2018-12-16:
@MabezDev You can use `#[rustfmt_skip]` (or something like that; I forget the correct name) to prevent rustfmt from reformatting that line.

> does this need to be fixed in the macro parsing?

We can't do anything from our side. If you hit the compiler bug then all span information provided by the compiler to the macro is incorrect.

MabezDev, 2018-12-16:
>You can use #[rustfmt_skip]

Thanks, that's what I'll do for now, in the mean time I've opened an [issue on rust-lang/rust](https://github.com/rust-lang/rust/issues/56894)

closed 2018-12-16 by MabezDev