rtic-rs/rtic #104: Multi-locks

japaric, 2018-11-14:
AKA locking several resources at once.

Mock API:

``` rust
(resources.A, resources.B).lock(|a: &mut u32, b: &mut i32| { .. })
// repeat for tuples of other sizes
```

Benefits compared to

``` rust
resources.A.lock(|a| {
    resources.B.lock(|b| {
        // ..
    });
});
```

- Less rightward drift
- Only a single critical section is created
- All resources are *atomically* locked (locked "at the same time")

Requirements:

- This should not complicate generic code. So no more user facing traits should be added.

- Also, ideally, this should Just Work:

``` rust
fn foo(a: impl Mutex<T = u32>, b: impl Mutex<T = i32>) {
    a.lock(|a| {});
    b.lock(|b| {});
    (a, b).lock(|a, b| ..)
}
```

arkap, 2019-03-06:
I was just going to open an issue on this topic, because when I try to acquire multiple locks like in your example:

> ```
> resources.A.lock(|a| {
>    resources.B.lock(|b| {
>        // ..
>    });
> });

the borrow-checker complains about multiple mutable borrows of `resources`.
So yeah, I would love to see a possibility to lock multiple resources atomically.

TeXitoi, 2019-03-06:
Maybe this is working:
```rust
let b = &mut resources.B
resources.A.lock(|a| {
   b.lock(|b| {
       // ..
   });
});
```

TeXitoi, 2019-03-06:
@arkap https://stackoverflow.com/questions/36379242/mutably-borrow-one-struct-field-while-borrowing-another-in-a-closure

arkap, 2019-03-06:
Yes, that did the trick. Thank you!
I would still love to see an explicit API for this, though. Makes for more readable code.

TotalKrill, 2019-10-30:
Yes, this is an annoyance when using RTFM, The rightward drift is hard to read and reason about, and doesn't give much benefit. Would love to see this work

korken89, 2019-12-29:
This feature comes with [`mutex-trait`](https://github.com/rust-embedded/mutex-trait) from the embedded WG. Closing

closed 2019-12-29 by korken89

little-arhat, 2020-01-08:
@korken89 any examples or docs on how to integrate it with RTFM? Does RTFM analysis "knows" about this mutex trait?