rtic-rs/rtic #105: ARMv6: use interrupt masking instead of "global" critical sections

japaric, 2018-11-14:
Right now `Mutex.lock` is disabling *all* interrupts when compiled for Cortex-M0(+)

korken89, 2019-12-29:
This should be moved into an RFC, can you fix one @japaric ?

japaric, 2020-01-06:
I believe @perlindgren and / or @nils-grepit have been working on this. They can probably submit a RFC / PR.

One unresolved question that I didn't mention in the issue description is: what is the expected behavior of the user masking an interrupt within a `lock` closure that also masks that interrupt? See code below:

``` rust
#[task(binds = A, resources = [x], priority = 1)]
fn foo(cx: foo::Context) {
    cx.x.lock(|_| { // interrupt B masked
        NVIC::mask(Interrupt::B);
    }); // interrupt B unmasked
    // interrupt B is unmasked at this point but that's probably NOT what the user wanted
}

#[task(binds = B, resources = [x], priority = 2)]
fn bar(cx: bar::Context) {
    // ..
}
```

japaric, 2020-01-06:
This issue is not actionable without a RFC that answers the above question and gives more details about how the masking will be done so I'm going to close this in favor of a proper RFC, or just a thread to start, in rtfm-rs/rfcs

closed 2020-01-06 by japaric