rtic-rs/rtic #106: Problems with interrupt attribute with stm32f7/stm32f7x7

N5FPP, 2018-11-26:
When trying to compile example/interrupts.rs I get the error below about the 'interrupt' attribute not being known. I am testing against 0.4.0-beta.2. The only differences to the original code is that I'm using the stm32f7 and my own stm32f7x7 crates for my device (diff at the bottom).  

Is 0.4.0-beta.2 compatible with theses crates?  

Due to a warning/errors about _private_no_mangle_statics_ and _macro_reexport_ no longer being available in my nightly build (1.32.0-nightly), I have tried running _svd2rust_ to generate my own stm327x7 locally and get the exact same error. 

I have been able to build and run examples/init.rs all the way out to my NUCLEO-F767ZI dev board.

-Freeman
 
`
pascal@tsunami  (stm32f7x7) <51> cargo build --example interrupt
   Compiling proc-macro2 v0.4.24
   Compiling semver-parser v0.7.0
   Compiling unicode-xid v0.1.0
   Compiling rand_core v0.3.0
   Compiling typenum v1.10.0
   Compiling cortex-m v0.4.3
   Compiling vcell v0.1.0
   Compiling cortex-m-rt v0.6.5
   Compiling cortex-m v0.5.8
   Compiling aligned v0.2.0
   Compiling untagged-option v0.1.1
   Compiling bare-metal v0.1.3
   Compiling aligned v0.1.2
   Compiling cortex-m-semihosting v0.3.2
   Compiling stm32f7 v0.4.0
   Compiling byteorder v1.2.7
   Compiling r0 v0.2.2
   Compiling cortex-m-rtfm v0.4.0-beta.2 (/home/pascal/Workspaces/Rust/cortex-m-rtfm)
   Compiling stable_deref_trait v1.1.1
   Compiling panic-halt v0.2.0
   Compiling volatile-register v0.2.0
   Compiling rand_core v0.2.2
   Compiling rand v0.5.5
   Compiling semver v0.9.0
   Compiling hash32 v0.1.0
   Compiling stm32f7x7 v0.2.0
   Compiling rustc_version v0.2.3
   Compiling bare-metal v0.2.4
   Compiling quote v0.6.10
   Compiling generic-array v0.11.1
   Compiling generic-array v0.12.0diff --git a/examples/interrupt.rs b/examples/interrupt.rs
index 3c669d9..9f63429 100644
--- a/examples/interrupt.rs
+++ b/examples/interrupt.rs
@@ -6,12 +6,14 @@
 #![no_std]
 
 extern crate panic_semihosting;
+extern crate stm32f7;
 
 use cortex_m_semihosting::{debug, hprintln};
-use lm3s6965::Interrupt;
+use stm32f7::stm32f7x2;
+use stm32f7x7::Interrupt;
 use rtfm::app;
 
-#[app(device = lm3s6965)]
+#[app(device = stm32f7x7)]
 const APP: () = {
     #[init]
     fn init() {

   Compiling panic-semihosting v0.5.1
   Compiling syn v0.15.22
   Compiling heapless v0.4.0
   Compiling as-slice v0.1.0
   Compiling cortex-m-rt-macros v0.1.3
   Compiling owned-singleton-macros v0.1.0
   Compiling cortex-m-rtfm-macros v0.4.0-beta.2 (/home/pascal/Workspaces/Rust/cortex-m-rtfm/macros)
   Compiling owned-singleton v0.1.0
   Compiling alloc-singleton v0.1.0
**error[E0658]: The attribute 'interrupt' is currently unknown to the compiler and may have meaning added to it in the future (see issue #29642)**
  --> examples/interrupt.rs:16:1
   |
16 | #[app(device = stm32f7x7)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = help: add #![feature(custom_attribute)] to the crate attributes to enable

error: aborting due to previous error

For more information about this error, try 'rustc --explain E0658'.
error: Could not compile 'cortex-m-rtfm'.

To learn more, run the command again with --verbose.
`
diff --git a/examples/interrupt.rs b/examples/interrupt.rs
index 3c669d9..9f63429 100644
--- a/examples/interrupt.rs
+++ b/examples/interrupt.rs
@@ -6,12 +6,14 @@
 #![no_std]
 
 extern crate panic_semihosting;
+extern crate stm32f7;
 
 use cortex_m_semihosting::{debug, hprintln};
-use lm3s6965::Interrupt;
+use stm32f7::stm32f7x2;
+use stm32f7x7::Interrupt;
 use rtfm::app;
 
-#[app(device = lm3s6965)]
+#[app(device = stm32f7x7)]
 const APP: () = {
     #[init]
     fn init() {

geomatsi, 2018-11-26:
Reference examples for lm3s6965 are supposed to be run with QEMU. IIUC device crates are not yet ready for using with up-to-date RTFM. It will be possible to use current RTFM after svd2rust 0.14 is released and device crates are converted accordingly.

aurelj, 2018-12-05:
You should be able to get you stm32f7x7 working with rtfm 0.4.0 beta already, but you will need to generate your stm32f7x7 with the current git version of svd2rust:
```
cargo install svd2rust  --git https://github.com/rust-embedded/svd2rust.git
```

You also need to enable the "rt" feature of the stm32f7 crate in your Cargo.toml to ensure the 'interrupt' macro is exported:
```
[dependencies.stm32f7]
version = "0.4.0"
path = "../stm32-rs/stm32f7"
features = ["rt", "stm32f7x7"]
```

DoumanAsh, 2018-12-07:
New svd has been released so it can be used to update memory map too

japaric, 2018-12-16:
As other have indicated you have to use a peripheral access crate (PAC) that's been generated using svd2rust v0.14.x. If you encounter issues using such a PAC then let us know.

closed 2018-12-16 by japaric