rtic-rs/rtic #109: Scheduled task not executing at all on STM32F0

aurelj, 2018-12-05:
When running examples/schedule.rs with v0.4.0-beta.3 on STM32F0x0 (using the stm32f0 crate rebuilt using git bersion of svd2rust) the tasks do not run at all.

Other rtfm examples (not using timer-queue) are running fine on this STM32F0. And this schedule.rs example is running perfectly on a STM32F4, but on a STM32F0, here is the only output I get:
```
init @ Instant(0)
```

Can it be related to some ARMv6 vs. ARMv7 issue ?
I have no idea where to look at first to identify the issue.

aurelj, 2018-12-09:
I've found the root of this issue: Instant::now() always return 0 on stm32f0.

This is due to the DWT CYCCNT register not being available on stm32f0 (and IIUC on any Cortex-M0).
So basically, timer-queue won't work on stm32f0 for now.

I'm wondering if we could use the SysTick counter along with a SysTick accumulator updated by the SysTick ISR, to implement Instant::now() on hardware which don't have a usable DWT.

ceciliacsilva, 2018-12-14:
I'm having the same issue with my stm32l0 (cortex-m0+), everything works except timer queue... This crate calls `DWT::get_cycle_count()` that calls `(*Self::ptr()).cyccnt.read()` not available in ours platforms. 

Looking further, the entire DWT module does not seem to match what's expected for my processor m0+ (http://infocenter.arm.com/help/topic/com.arm.doc.ddi0484c/DDI0484C_cortex_m0p_r0p1_trm.pdf) (https://github.com/rust-embedded/cortex-m/blob/39172497729e827a6bb48f899a1facf5ceb2558b/src/peripheral/dwt.rs). It is m3-m4 compatible m3(https://static.docs.arm.com/ddi0337/h/DDI0337H_cortex_m3_r2p0_trm.pdf), m4 (http://infocenter.arm.com/help/topic/com.arm.doc.ddi0439b /DDI0439B_cortex_m4_r0p0_trm.pdf), I did not check but I believe that m7 too. In the current version of 'cortex-m', I did not see any compilation flag in the 'peripherals' module handling this case.

closed 2018-12-16 by japaric