rtic-rs/rtic #115: Support device-specific timer queue implementations

japaric, 2018-12-16:
The current timer queue implementation uses Cortex-M peripherals (DWT and SysTick) available on ARMv7-M+. In theory, it should be possible to use device specific peripherals to implement the timer queue. This issue is for tracking progress on implementing that feature (user defined timer queue peripherals).

ceciliacsilva, 2018-12-17:
I believe this can be useful for those who wish to know a little more about the theme: http://ceur-ws.org/Vol-1464/ewili15_16.pdf (theoretically).

eupn, 2019-02-03:
@japaric do you have in mind an API for this? If you describe it I probably could implement it as a PR.

japaric, 2019-02-08:
No, I haven't really thought about this. There's a design question about how to initialize the user-defined timer queue, which right now has to be already initialized by the start of `init` (you can use the spawn API in init) -- this might require some changes in the macros.

The other question is what should be the API of the trait that timers that can be used with a timer queue have to implement. This likely involves a user defined `Duration` (e.g. 64-bit integer under the hood), which would have to become an associated type of the "timer" trait.

korken89, 2019-12-29:
This should probably become an RFC for proper discussion

japaric, 2020-01-06:
> This should probably become an RFC for proper discussion

Indeed. rtfm-rs/rfcs is the right place to discuss this. We can discuss in rtfm-rs/rfcs#23 and / or rtfm-rs/rfcs#10 so I'm going to close this issue.

closed 2020-01-06 by japaric