rtic-rs/rtic #117: Use of SYST within #[init] with timer-queue

MabezDev, 2018-12-17:
I've finished porting [my project](https://github.com/MWatch/mwatch/tree/rtfm-v4-hal) to v4, overall very easy to switch to the new attribute system.

During init I use the systick delay quite a lot, to reset displays, wait for devices to boot etc. I am wondering if the timer-queue is not using systick during init, could this be used by the user, then passed back somehow?

japaric, 2018-12-18:
> During init I use the systick delay 

Note that there is also [asm::delay](https://docs.rs/cortex-m/0.5.8/cortex_m/asm/fn.delay.html) which should be cycle accurate in `init` since interrupts are disabled (but beware of non-zero Flash latency).

>  I am wondering if the timer-queue is not using systick during init, could this be used by the user, then passed back somehow?

Actually, I think we can let you borrow `SYST` during `init` even if you do use the `schedule` API in `init`. Though the required changes are not straightforward.

japaric, 2018-12-18:
>  let you borrow SYST during init

and by this I mean that `SYST` would be available as a field of `rtfm::Peripherals` even when `timer-queue` is enabled.

perlindgren, 2018-12-18:
An idea for the future is to implement multiple operation modes in RTFM,  (e.g., besides init, we could have startup, run, low-power, etc. each having a different sets of tasks and resources and capabilities, e.g., clock frequency, clocked peripherals etc.). The question is how much support would we want to be built into the RTFM framework, (besides the ceiling analysis/resource protection, message queues and timer tasks (dispatching the postponed messages).

Regarding the semantics, one could think of different ways to handle mode changes/transitions (if modes involves changing of clock trees, etc. it becomes more involving).

A major concern here is that RTFM is a hard-real time system aiming at static guarantees to robustness and timing properties. In consequence, at any given point in time, it is essential that the system remains amenable to analysis (e.g., response times, stack-memory etc.) even during mode transitions.

Best regards
  Per

MabezDev, 2018-12-18:
> and by this I mean that SYST would be available as a field of rtfm::Peripherals even when timer-queue is enabled.

Yeah I think this would be nice. What are the complications of allowing syst in the rtfm peripherals?
> Note that there is also asm::delay which should be cycle accurate

Thanks I didn't know about `asm::delay`. The crates I'm using require a delay object that implemented `embbedded_hal::blocking::delay::DelayMs`, I suppose I could wrap `asm::delay` in a struct and impl `DelayMs` on it to sort that issue.

little-arhat, 2019-03-06:
Just hit the same issue, would be nice to have embbedded_hal::blocking::delay::DelayMs provided by rtfm, as many driver crates require that for initialization.

japaric, 2019-03-07:
@little-arhat it can't because RTFM is a device agnostic framework and translating from milliseconds to clock cycles requires knowing the clock frequency, which today requires a device-specific API. Also I don't really see the need for it; most HAL crates provide implementations of `DelayMs`.

It would be nice to have [a device-agnostic API to access the core clock frequency](https://github.com/rust-embedded/embedded-hal/issues/129) though; we could use that to provide a variation of the `schedule` API that works with time units like milliseconds instead of clock cycles.

little-arhat, 2019-03-07:
@japaric it doesn't have to be automatically configured, at the very least rtfm can require user to pass clock frequency to Delay struct constructor, so they wouldn't have to write [asm::delay based implementations](https://github.com/copterust/asm-delay) themselves.

mjadczak, 2019-03-11:
> translating from milliseconds to clock cycles requires knowing the clock frequency, which today requires a device-specific API

Does [`SYST::get_ticks_per_10ms`](https://docs.rs/cortex-m/0.5.8/cortex_m/peripheral/struct.SYST.html#method.get_ticks_per_10ms) not allow us to get the time <-> systick mapping? Or is that value not reliable enough?

japaric, 2019-05-02:
@mjadczak that calibration value is not always present (e.g. nRF52840 doesn't have it) and when present it doesn't seem to always be correct (e.g. the STM32F103 reports a value of `9000` but the correct value is `7999`) or follow ARM specification (the vendor is allowed to give the register a different meaning).

korken89, 2019-12-29:
This seems to be solved.

closed 2019-12-29 by korken89

justacec, 2020-04-04:
How was this solved?

WyseNynja, 2020-07-18:
I'm also wondering how this is solved.

Before rtic, I was doing `let delay = feather_m0::delay::Delay::new(device.SYST, &mut clocks);`, but I don't know how to get SYST since it isn't a valid field. Is there an alternative?

korken89, 2020-09-15:
Hi, sorry for missing this @justacec and @WyseNynja .

How to get around this there are 2 direct options,

1. Do not use monotonic and you still have access to the SysTick.
2. Make a PR to the HAL/BSP with an alternative implementation, e.g. based on `cortex_m::asm::delay` or another hardware timer.