rtic-rs/rtic #121: Instant::elapsed() may give incorrect Duration values

jamesmunns, 2018-12-25:
As `Instant`s are created using solely the DWT, they are limited to a range of 32 bits. For example, the nRF52832 has a fixed system clock of 64MHz. This means that the cycle count will roll over every 67 seconds or so.

If you do the following:

```rust
let start = Instant::now(); // Lets say this happens at cycle 0xFFFF_FFF0
// ... some time passes
let elapsed = start.elapsed(); // Lets say this happens just after the rollover, at the next 0x0000_000F

// `elapsed` will now be (0x0000_000F - 0xFFFF_FFF0)
// which in debug mode *should* panic, and in release mode will == 31, which is correct

// Now we wait 670 seconds or so, when 10 more rollovers have happened
// timer is now 0x0000_000F after all of those rollovers
let elapsed_2 = start.elapsed();

// `elapsed` will now be (0x0000_000F - 0xFFFF_FFF0)
// which is still `31`, and incorrect!
```

Many devices have a much faster core clock than 64MHz, encountering this more often.

The larger issue is a case of aliasing. Let's say you are writing a debounce algorithm, which fires on every rising OR falling edge of a button press.

```rust
match state {
    RisingEdge => {
        *TRIGGER_TIME = Instant::now();
    }
    FallingEdge => {
        if *TRIGGER_TIME.elapsed() >= 16_000_000.cycles() { // 250ms at 64MHz
             do_something();
        } else {
             // do nothing, bouncy trigger
        }
    }
}
```

Here's what we would get, depending on the `cycle_count` time, if the RisingEdge occurred at `0x8000_0000`:

| Time (cycles) at Falling Edge | Duration (cycles) since Rising Edge | Behavior | Correct? |
| :---                                       | :---                                            | :---         | :---         |
| 0x80f4_23ff | 15_999_999 | do nothing | YES |
| 0x80f4_2401 | 16_000_001 | do_something() | YES |
| 0xffff_ffff | 2_147_483_647 | do_something() | YES |
| 0x0000_0000 (+1 rollover) | 2_147_483_648 | do_something() | YES |
| 0x7f0b_dc00 (+1 rollover) | 4_278_967_296 | do_something() | YES |
| 0x8000_0001 (+1 rollover | 1 | do nothing | NO |

japaric, 2019-05-08:
> Let's say you are writing a debounce algorithm

That's out of scope for the `rtfm::{Instant,Duration}` APIs. I'm documenting the scope and limitations of the schedule, Instant and Duration APIs in #190.

korken89, 2019-12-29:
This is "solved" as per above

closed 2019-12-29 by korken89