rtic-rs/rtic #122: Empty idle loop causes hard fault

zklapow, 2018-12-25:
I was working on a project the other day and evenually my debug builds started to become too large for the tiny microcontroller I was running on (an STM32f0706p6 if that is relevant), so I decided to try setting `opt-level = "z"` but when I did this I immediately started getting a hard fault on startup. I tried messing with all sorts of different settings (LTO vs no LTO, LTO in combo with different opt-levels, release vs debug builds) but eventually determined that anything above `opt-level = 0` caused the fault.

On a hunch I removed the empty idle loop I had in my app and voila the issue disappears. Here is a very simple test app which reproduces the problem:

```rust
#![no_std]
#![no_main]

extern crate panic_semihosting;

use cortex_m_semihosting::{debug, hprintln};
use rtfm::app;

use stm32f0::stm32f0x0;

#[app(device = stm32f0::stm32f0x0)]
const APP: () = {
    static mut COUNTER: u16 = 0;

    #[init(spawn = [foo])]
    fn init() {
        spawn.foo().unwrap();
    }

    #[idle]
    fn idle() -> ! {
        loop {
        }
    }

    #[task(resources = [COUNTER])]
    fn foo() {
        static mut ONCE: bool = true;

        hprintln!("foo: {:?}", *resources.COUNTER).unwrap();

        *resources.COUNTER += 1;
    }

    extern "C" {
        fn TIM16();
    }
};
```

Compiled in debug mode this just loops in idle forever, compiled in release mode this hard faults. If you modify idle to look like:

```rust
    #[idle()]
    fn idle() -> ! {
        loop {
            hprintln!("idle");
        }
    }
```

Everything works.

 My hunch is that RTFM is somehow referencing the idle function still but that it is getting optimized out. I may go digging into the RTFM source to see if I cant solve this myself but figured I would post here first in case anyone had an easy fix for the problem.

Obviously the easiest solution is just not to have an empty loop in idle but this seems like a less than ideal user-experience as it is a relatively subtle and hard to debug state.

perlindgren, 2018-12-25:
Hi,

This issue is not a problem of RTFM or even Rust per-se, its a problem of LLVM. Its allowed to optimize loop {} into UB, unless the loop has a side effect, an asm nop will do or some atomic/fence operator. There is currently no satisfying solution to that matter (other than as you already found out, do "something" in the loop. If you don't give an idle loop at all there should  be a default idle with correct behavior implemented by the framework.

(Problem known forever, aka 2015, https://github.com/rust-lang/rust/issues/28728)

Best regards

 Per

kuon, 2020-09-14:
I hit the same issue today, but it took me quite a long time to figure out.

I think it should be mentioned in the documentation, here:

https://rtic.rs/0.5/book/en/by-example/app.html#idle

Add something like:

"If you specify an empty idle function, and use a build profile with some optimization level, it will be optimized away and cause a crash. Be sure to either have a non empty idle function or no function at all."

korken89, 2020-09-15:
Thanks for the suggestion @kuon ! I'll make a PR highlighting this issue in the docs.

perlindgren, 2020-09-15:
You may opt to implement custom power save features, triggered when idle. If you don't give any idle, it will default to wfi if I recall correctly. 

We don't have any built in power features as of yet. The problem is closely related to modes of operation, where we might wish to have multiple task sets supported, e.g., a set of task triggered by an RTC calendar event (for doing some simple data acqusition, and store in NVM), another set of tasks where the system wakes the radio to transmit data, and
 things like that.

AfoHT, 2020-09-22:
Implemented in #360