rtic-rs/rtic #123: Timer-Queue does not seem to execute tasks when debugger is not attached

jamesmunns, 2018-12-28:
I have some examples which do the following:

* Toggle one LED periodically, based on TimerQueue Intervals - [code here](https://github.com/braun-robotics/rust-dwm1001/blob/2e194e427aae5a60bf443800d68e4a1eb478ada0/templates/rtfm-hal/src/main.rs#L119-L159)
* Toggle another LED when a button is pressed, driven by interrupts - [code here](https://github.com/braun-robotics/rust-dwm1001/blob/2e194e427aae5a60bf443800d68e4a1eb478ada0/templates/rtfm-hal/src/main.rs#L170-L231)

When the debugger IS attached, both of these features work correctly. When a debugger is NOT attached, only the interrupt driven task functions correctly. I suspect that without the debugger attached, certain trace functionality is not available, preventing the timer from firing correctly. I saw some discussion of this on [stack overflow](https://stackoverflow.com/questions/36378280/stm32-how-to-enable-dwt-cycle-counter), which might be interesting.

This is likely to be resolved when #115 lands.

TeXitoi, 2018-12-28:
Looks similar to https://github.com/rust-embedded/cortex-m/pull/111

TeXitoi, 2018-12-28:
Seems to be already called in RTFM, but not the access magic described on the stack overflow thread.

jamesmunns, 2018-12-28:
Possibly relevant (in case different vendors implement this functionality in different ways) - the specific chip in question here is the Nordic nRF52832, an `thumbv7em-none-eabihf` target

eupn, 2019-02-01:
Confirm, on my nRF52-DK (PCA10040) RTFM-based blinky example doesn't work without debugger attached.

japaric, 2019-02-08:
> It seems that either `DWT` cycle counter or `SYST` is disabled

AFAIS in stm32 land, these are behind device specific debug settings. stm32 uses, for example, the DBG / DBGMCU peripheral to set these settings. We can't do device specific stuff within rtfm so we can't fix this in rtfm. If you change the debug settings (if possible at all) in `init` then the timer queue will work.

perlindgren, 2019-02-08:
Hi

I think we might want to have an RTFM trait for enabling the DWT, that can be implemented for each family of MCUs. What do you think about that?

(There might be other RTFM traits later, for device specific timers, if we want to reduce priority inversion for the timer queue, and dispatch.)

Best

  Per

japaric, 2019-05-01:
People observing the timer queue not working / the DWT CYCCNT register not increasing: please try PR #185 or #186 and report back if that fixes this issue for you.

japaric, 2019-05-02:
> People observing the timer queue not working / the DWT CYCCNT register not increasing

I checked a nRF52840 and the "software lock" feature doesn't seem to be implemented for the DWT (register DWT_LSR reads as zero) -- the CYCCNT increases as expected -- so those PRs will likely not help for that chip. Perhaps there's some device / nRF specific lock that keeps the DWT disabled on reset (its clock could be disabled on reset to save power?).

korken89, 2019-12-29:
In v5 it is up to the user to initialize the monotonic timer.

closed 2019-12-29 by korken89