rtic-rs/rtic #126: [RFC] Message<T>: avoid copying message payloads

japaric, 2019-01-11:
## Summary

Add an opt-in mechanism to avoid copies when receiving messages.

## Background

Message passing always involves a copy on the receiver side. Internally, the
message payload is stored in a `static mut` variable. Right before the receiver
task starts this payload is copied (usually onto the stack) using the
`ptr::read` function to make the payload available *by value*.

This copy can be unnecessary when the task only needs a reference to the
payload. For example:

``` rust
#[rtfm::app]
const APP: () = {
    // ..

    // `packet` will be copied from a `static mut` variable onto `foo`'s stack
    #[task]
    fn foo(packet: [u8; 128]) {
        process(&packet);
    }
};

fn process(packet: &[u8]) {
    // ..
}
```

## Design

This RFC proposes extending the syntax of `#[task]` to provide an opt-in
mechanism to avoid these copies.

When the user specifies `x: rtfm::Message<T>` as the input of the `#[task]` the
codegen pass will avoid copying the payload; instead, the task will receive a
proxy to the payload: `rtfm::Message<T>`. This proxy only allows access *by
reference* to the payload.

The API of `rtfm::Message` is shown below:

``` rust
// crate: rtfm

pub struct Message<T> { /* .. */}

impl<T> ops::Deref for Message<T> {
    type Target = T;

    // ..
}

impl<T> ops::DerefMut for Message<T> {
    // ..
}

impl<T> Message<T> {
    /// Copies the message payload to the stack, freeing the message buffer
    pub fn read(self) -> T {
        // ..
    }
}

impl<T> Drop for Message<T> { /* .. */ }

// Sending `Message<T>` to a higher priority task would break the `Drop` impl
impl<T> !Send for Message<T> { /* ... */ }
```

Here's an updated version of our opening example. It avoids copying `packet`
onto `foo`'s stack.

``` rust
#[rtfm::app]
const APP: () = {
    // ..

    #[task]
    fn foo(packet: Message<[u8; 128]>) {
        process(&packet);
    }
};

fn process(packet: &[u8]) {
    // ..
}
```

It's important to note that using `rtfm::Message` will delay freeing the message
buffer -- remember that RTFM message queues have fixed capacity. This affects
the success of the `spawn` and `schedule` operations.

When `fn task(x: T)` is used the message buffer is freed *before* the task is
started, but when `fn task(x: rtfm::Message<T>)` is used the message buffer is
freed when `x` is dropped / destroyed. The code below illustrates the
difference.

``` rust
#[rtfm::app]
const APP: () = {
    #[init(spawn = [foo, bar, baz])]
    fn init() {
       // these never panic
       spawn.foo(0).unwrap();
       spawn.bar(1).unwrap();
       spawn.baz(2).unwrap();
    }

    #[task(capacity = 1, spawn = [foo])]
    fn foo(x: u32) {
        spawn.foo(x + 1).unwrap(); // this *never* panics
    }

    #[task(capacity = 1, spawn = [bar])]
    fn bar(x: Message<u32>) {
        spawn.bar(x + 1).unwrap(); // this *always* panics

        drop(x); // message buffer is freed here
    }

    #[task(capacity = 1, spawn = [baz])]
    fn baz(x: Message<u32>) {
        drop(x); // message buffer is freed here

        spawn.baz(x + 1).unwrap(); // this *never* panics
    }
};
```

### Why `rtfm::Message<T>` and not `Message<T>`?

Using the latter syntax would break code like this:

``` rust
// User defined struct
struct Message<T> { /* .. */ }

#[rtfm::app]
const APP: () = {
    // ..

    #[task]
    fn foo(x: Message<u32>) {
        bar(x);
    }
};

fn bar(x: Message<u32>) {
    // ..
}
```

The prefix is needed to avoid breaking existing code.

## Benchmark

Consider the following program:

``` rust
#![deny(warnings)]
#![no_main]
#![no_std]

extern crate panic_halt;

use core::ptr;

use cortex_m::asm;

#[rtfm::app(device = stm32f103xx)]
const APP: () = {
    #[init(spawn = [foo])]
    fn init() {
        core.DWT.enable_cycle_counter();

        spawn.foo([0; 128]).ok().unwrap();
    }

    #[task]
    fn foo(x: [u8; 128]) {
        let sum: u32 = x.iter().cloned().map(u32::from).sum();
        unsafe {
            ptr::read_volatile(&sum);
        }
        drop(x);
        asm::bkpt();
    }

    extern "C" {
        fn EXTI0();
    }
};
```

It takes 778 cycles to get to `asm::bkpt` from `EXTI0`. If the signature of
`foo` is changed to `fn(rtfm::Message<[u8; 128]>)` to avoid copying the array
then the execution time reduces to 421 cycles (~46% reduction).

## Alternatives

We could require that the input of `#[task]`s is *always* `Message<T>`. This
is a breaking change because the signature of all existing `#[task]`s would
have to change from `fn(a: A, b: B, ..)` to `(msg: Message<(A, B, ..)>)`.
Also, it's not always beneficial to "avoid the copy". In the case of small
values like `u32` the payload will likely be loaded into a register instead of
being copied onto the stack. The main proposal is more flexible than this
alternative.

## Unresolved questions

- Can this approach be used on the sender side to avoid copies from the stack to
the `static mut` message buffer? For example, an operation like `spawn.foo([0;
128])` will allocate an array on the stack, zero it (memclr) and then copy
(memcpy) it into the message buffer (`static mut` variable).

We could use a newtype over `MaybeUninit<T>` (`Envelope<T>`?) that `DerefMut`s
to `MaybeUninit<T>`. (A.N. This seems like it would deserve its own RFC /
discussion thread).

- Bikeshed the name of `rtfm::Message`.

---

There's a reference implementation in #XXX, but note that, at the time of
writing, the implementation uses `Message<T>` instead of `rtfm::Message<T>` --
this needs to be fixed to fully adhere to the RFC.

korken89, 2019-02-16:
Overall I see the use of this and approve, especially in RF system where passing around a large buffer is common.
Is there something specific that talks against this? Seems like only + for me.

japaric, 2019-03-07:
My issue with this API is that is rather limiting. You can't send a `Message` to
another task because it has non-static lifetime (it's tied to the receiver
task). Also while you hold the `Message` you also hold a spot in the message
queue meaning that, for example, if the task capacity is one another message
can't be enqueued until `Message` is dropped. Finally, this proposal doesn't
address the memcpy on the sender side.

I think that a (lock-free) memory pool is a much better solution to the copying
problem. [`lifo::Pool`], which I came up with after posting this proposal,
solves the problem today (it works with RTFMv4) and it's ergonomic to use.

[`lifo::Pool`]: https://japaric.github.io/lifo/lifo/index.html

I haven't published any elaborated example using `lifo::Pool` but here's a small
example that shows how it avoids copying both on the sender and receiver sides.

``` rust
// yes, declared outside `APP`; the pool doesn't depend on RTFM -- it's lock-free
pool!(A: [u8; 128]);

#[app(device = /* .. */)]
const APP: () = {
    #[interrupt(binds = EXTI0, resources = [RADIO], spawn = [process])]
    fn on_new_packet() {
        // proxy for the new pakect; only contains the length of the packet
        let pending_packet = RADIO.pending_packet();

        if let Some(buffer) = A::alloc() { // alloc takes ~10 clock cycles
            // `buffer` has type `Box<A, Uninitialized>` and it's just a pointer

            // leaves the memory uninitialized; this is a no-op
            // see https://github.com/rust-lang/rust/pull/58363 for details
            // (the alternative is zeroing the buffer but that's an O(N) operation)
            let buffer: Box<A> = buffer.freeze();

            // `OwningSliceTo` is basically `slice[..end]` but by value
            let packet: OwningSliceTo<Box<A>, u8> = pending_packet.read(buffer);

            // this copies only the pointer and the `end` index (`u8`)
            // the contents of the buffer / packet are *not* copied
            spawn.process(packet).expect("UNREACHABLE");
        } else {
            warn!("no space to copy the new packet; dropping it");
            pending_packet.discard();
        }
    }

    #[task]
    fn process(packet: OwningSliceTo<Box<A, u8>) {
        // on the receiver side we only copy the pointer and the `end` index
        // again, the contents of `packet` are *not* copied

        // the slot in the message queue has been freed at this point and another
        // message can be enqueued

        // .. do stuff with `packet` ..

        // eventually (this happens automatically) the memory is returned to the pool
        drop(packet); // this takes ~10 clock cycles
    }
};
```

korken89, 2019-03-12:
Ah, this is indeed a good way to do it!
I would add this as a pattern and not accept this RFC.

japaric, 2019-04-22:
I have made a PR to document how to use `heapless::Pool` to make message passing of buffers faster. #179

korken89, 2019-04-28:
With the `pool` example I vote for closing this RFC.

japaric, 2019-05-02:
Let's formally vote on **rejecting** this RFC:

- [x] @japaric 
- [x] @korken89 
- [ ] @TeXitoi 

We have majority but let's wait a few days to see if anyone raises a concern.

TeXitoi, 2019-05-02:
`heapless::Pool` seems a better solution to me, even if I didn't dive to deep in this subject.

japaric, 2019-05-08:
About a week has passed and no one has raised a concern so I'm going to formally close / reject this RFC.

commit 2451e07c (2019-04-28): Merge #179

179: [backport] book: indirection for faster message passing r=japaric a=japaric

closes #126 

documents how to use indirection, specifically `heapless::Pool`, to reduce the overhead of message passing.
PR #179 (2019-05-01): [backport] book: indirection for faster message passing
closes #126 

documents how to use indirection, specifically `heapless::Pool`, to reduce the overhead of message passing.
closed 2019-05-08 by japaric