rtic-rs/rtic #128: [RFC] #[interrupt(binds = "..")]

japaric, 2019-01-11:
## Summary

Extend the syntax to let the user name interrupt and exception tasks differently
from the interrupt / exception they bind to.

## Background

This code doesn't compile today:

``` rust
//~ ERROR: the name `USART1` is defined multiple times

#![no_main]
#![no_std]

extern crate panic_halt;

use stm32f103xx::USART1;

#[rtfm::app(device = stm32f103xx)]
const APP: () = {
    static mut USART: Option<USART1> = None;

    #[init]
    fn init() {}

    #[interrupt]
    fn USART1() {}
};

```

One workaround is to remove the `USART1` import and use `stm32f103xx::USART1`
instead of `USART1` in the definition of the `USART` resource.

The underlying issue is that the `#[interrupt]` attribute will expand into a
module named `USART1`. This module contains types like `crate::USART1::Spawn`
and `crate::USART1::Schedule` (note the `USART1` prefix) and it's required to
differentiate the `Spawn` struct of one task from the others'.

## Design

This RFC proposes extending the syntax of both the `#[exception]` and
`#[interrupt]` attributes to take a `binds` argument.

The value of `binds` must be a valid exception / interrupt name for the target
device. When omitted `binds` defaults to the task (function) name. The module
in the expansion of `#[exception]` / `#[interrupt]` will match the task and not
the `binds` value.

An example to illustrate the changes:

``` rust
// No import collision this time
use stm32f103xx::USART1;

#[rtfm::app(device = stm32f103xx)]
const APP: () = {
    static USART: USART1 = ();

    #[init]
    fn init() {
        // ..
    }

    #[interrupt(binds = USART1, resources = [USART], spawn = [foo])]
    fn serial() {
        bar(spawn);

        // ..
    }

    #[task]
    fn foo() {
        // ..
    }

    extern "C" {
        fn EXTI0();
    }
};

// NOTE: `serial::Spawn` and not `USART1::Spawn`
fn bar(_: serial::Spawn) {}
```

## Unresolved questions

- Bikeshed alternative names for `binds`.

TeXitoi, 2019-01-11:
It can also help with code readability: having `#[interrupt] fn UART1()` is not very explicit, but having `#[interrupt(binds = UART1)] fn command_channel()` is much more readable.

korken89, 2019-02-16:
I agree that this helps readability in code. 
Moreover I think that this would help make auditing code easier as good function names help solve misinterpretations.

japaric, 2019-02-19:
I think this RFC has been uncontroversial so I'm going to move it into the "accepted" bin and work on implementing in for one of the next v0.4.x releases.

korken89, 2019-02-21:
SGTM!

commit 6d1d8498 (2019-02-26): Merge #158

158: implement RFC #128: #[interrupt(binds = ..)] r=korken89 a=japaric

closes #128 

r? @korken89 or @TeXitoi 

suggestions for tests are welcome! (2 of the 3 tests I added hit bugs in my implementation)
PR #158 (2019-02-26): implement RFC #128: #[interrupt(binds = ..)]
closes #128 

r? @korken89 or @TeXitoi 

suggestions for tests are welcome! (2 of the 3 tests I added hit bugs in my implementation)
closed 2019-02-26 by bors[bot]