rtic-rs/rtic #130: Cannot use non-`Send` resource.

arkap, 2019-01-11:
In my project I have a resource `struct` with a `*mut`-member, so it's non-`Send`. I want a periodic task to use this resource. As of now it is the only task that needs access to it.

Compilation fails because of `rtfm::export::assert_send` although the [RTFM book] says that 
> you do not need the `Send` trait on types that are transferred between tasks that run at the same priority

Am I missing something or is this a bug?

[RTFM book]: https://japaric.github.io/cortex-m-rtfm/book/by-example/types-send-sync.html#send

TeXitoi, 2019-01-11:
This quote is about messages transfered to task, not resources.

You can use a `static mut` in your task to do the job. It will be trivial if you don't need the peripherals to construct the initial value.

Another solution is implementing the Send trait. You generally can do it if the field is private and only used in methods that takes `&mut self`.

TeXitoi, 2019-01-11:
Are you using this resource in `init`? Can you provide a minimal example not working?

arkap, 2019-01-11:
> Another solution is implementing the Send trait. You generally can do it if the field is private and only used in methods that takes `&mut self`.

This sentence in particular made me realize what Rust's ownership system does for thread-safety. Thanks, I learned something :)

TeXitoi, 2019-01-15:
I think that's a missing feature. @japaric I let you status on this one.

japaric, 2019-01-15:
This is pretty much intentional as it's required for memory safety.

Initializing a late resource is equivalent to moving a value into a new thread (e.g. `let x = ..; thread::spawn(move || /* use x here */)`) so the `Send` bound is required. Here's an example that shows how removing the bound could lead to UB.

``` rust
#[rtfm::app(device = stm32f103xx)]
const APP: () = {
    // NOTE: `Arc<T>` is only `Send` when `T` is `Sync`, and `RefCell<T>` doesn't impl `Sync`
    static mut A: Arc<RefCell<u32>> = ();
    static mut B: Arc<RefCell<u32>> = ();

    #[init]
    fn init() {
        let arc = Arc::new(RefCell::new(0));

        A = arc.clone(); // like spawn::EXTI0(move || /* move arc.clone() here */)
        B = arc.clone(); // like spawn::EXTI1(move || /* move arc.clone() here */)
    }

    #[interrupt(priority = 1, resources = [A])]
    fn EXTI0() {
        let a = resources.A.borrow_mut();

        rtfm::pend(Interrupt::EXTI1);
    }

    #[interrupt(priority = 2, resources = [B])]
    fn EXTI1() {
        // both `a` and `b` are mutable references to the same memory location
        // this breaks Rust aliasing model and it's UB
        let b = resources.B.borrow_mut();
    }
};
```

TeXitoi, 2019-01-15:
Maybe it need to be explained in the book then.

commit 59d8ef18 (2019-04-21): Merge #169

169: book: note that late resources must be Send r=japaric a=japaric

closes #130
commit 584483ab (2019-04-21): Merge #169

169: book: note that late resources must be Send r=japaric a=japaric

closes #130
PR #169 (2019-04-21): book: note that late resources must be Send
closes #130
closed 2019-04-21 by japaric