rtic-rs/rtic #131: Set device.DBG.cr.dbg_sleep to 1 when using wfi in the idle task.

nraynaud, 2019-01-15:
I was struggling to flash my device with openocd after I started using RTFM while googling I found this message:
https://www.jsykora.info/2014/10/error-jtag-status-contains-invalid-mode-value-communication-failure-solved/

adding this at the top of my init seems to have helped:
`    
	#[init(schedule = [toggle])] 
	unsafe fn init() {
        let dbg = device.DBG;
        dbg.cr.modify(|_, w| w.dbg_sleep().set_bit());
    }
`

nraynaud, 2019-01-15:
I am wrong, this has to be done in the debugger when we need it, this re-activate a clock.

japaric, 2019-02-08:
>  let dbg = device.DBG;

This is device specific and can't really be done in rtfm. It could be the case that the device does not have a DBG peripheral; or it may not be named like that; or its registers / bit fields may be named differently; or the svd file may not contain bitfield information.

nraynaud, 2019-02-08:
I see.

closed 2019-02-08 by nraynaud