rtic-rs/rtic #135: [RFC] Invoking software tasks

japaric, 2019-01-25:
Add the ability to start a task via a function call, removing the need for
sending message which results in lower overhead.

## DSL additions

All function attributes (e.g. `#[task]`, `#[interrupt]`) will accept a new,
named `call` argument. This argument is a list of software `#[task]`s that the
handler can directly start via a function call.

The example below showcases the new syntax

``` rust
#[rtfm::app(device = ..)]
const APP: () = {
    // ..

    #[task(call = [bar], spawn = [bar])]
    fn foo() {
        spawn.bar(1337);
        call.bar(42);
    }

    #[task]
    fn bar(x: u32) {
        println!("The answer is {}", x);
    }
};
```

Output:

``` console
The answer is 42
The answer is 1337
```

`call`-able tasks will appear as methods on a `call` value (`Call` struct).
`call`-ing a task starts it immediately; this instance of the task inherits the
priority of the caller.

### Restrictions

This new feature opens the door to many situations that could result in aliasing
of mutable references (`&mut`). This section describes the restrictions that
need to be placed on `call`-ed `#[task]`s to prevent that problem.

### Reentrancy

Because this allows more than one instance of a task to run concurrently the
task handler must be reentrant. This effectively means that the handler can
*not* contain task local variables. For instance, the program below would result
in mutable aliasing if it were accepted:

``` rust
#[rtfm::app]
const APP: () = {
    #[init(spawn = [foo])]
    fn init() {
        spawn.foo();
    }

    #[interrupt(priority = 2, call = [foo])]
    fn EXTI0() {
        call.foo();
    }

    #[task(priority = 1)]
    fn foo() {
        static mut STATE: u64 = 0;

        let state = &mut STATE;

        // this leads to UB
        rtfm::pend(EXTI0);
    }
};
```

EXTI0 preempts task `foo` and it invokes another instance of `foo`. This results
in two mutable references to `STATE` being created: one in the current `foo`
(current stack frame) and one in the previous `foo` (previous stack frame).

To prevent this issue the DSL will *reject* `call`-ing into tasks that contain
task local variables.

### Cycles

`call`-ing the same task in a cycle can result in mutable aliasing. Consider the
example below:

``` rust
#[rtfm::app(device = ..)]
const APP: () = {
    static mut SHARED: u64 = 0;

    #[init(spawn = [foo])]
    fn init() {
        spawn.foo();
    }

    #[task(resources = [SHARED], call = [foo])]
    fn foo() {
        resources.SHARED.lock(|shared: &mut u64| {
            if *shared == 0 {
                // cycle
                call.foo();
            }

            *shared += 1;

            // ..
        });
    }
};
```

When `SHARED.lock` is entered the second time two mutable references to `SHARED`
will exist: one in the current `foo` (current stack frame) and one in the parent
`foo` (previous stack frame). This breaks Rust aliasing rules.

To prevent this situation the DSL will forbid direct (e.g. `foo` -> `foo`) and
indirect (`foo` -> `bar` -> `foo`) `call` cycles.

### Resource overlap

Sharing resources between a task and its `call`-ee also results in mutable
aliasing. Consider the following program

``` rust
#[rtfm::app(device = ..)]
const APP: () = {
    static mut SHARED: u64 = 0;

    #[init(spawn = [foo])]
    fn init() {
        spawn.foo();
    }

    #[task(priority = 1, resources = [SHARED], call = [bar])]
    fn foo() {
        resources.SHARED.lock(|shared: &mut u64| {
            call.bar();
        })
    }

    #[task(priority = 2, resources = [SHARED])]
    fn bar() {
        let shared: &mut u64 = resources.SHARED;
    }
};
```

When `bar` is reached two mutable references to `SHARED` exist: one in the
current stack frame (`bar`) and one in the previous stack frame (`foo` handler).

To prevent this situation we need to forbid overlap of resources between a task
and its `call`-ee; this needs to consider both direct calls and indirect calls
(e.g. `foo` -> `bar` -> `baz`; there cannot be any resource overlap between
these three tasks).

## Changes in the ceiling analysis

The ceiling analysis pass must consider that `call`-able tasks will not only run
at their declared `priority`, but can also run at the priority of any task that
can `call` them.

## Drawbacks

This change requires adding non-trivial logic to the analysis pass, which
implies a significant implementation effort.

Several restrictions must be added to prevent mutable aliasing, which would
result in UB. This severely limits the number of tasks that are eligible for
being `call`-ed.

## Alternatives

The alternative is to *not* implement this. Users that want to both `call` and
`spawn` a task can instead refactor the common logic into a plain Rust function
and use the normal function call mechanism. For example:

``` rust
#[rtfm::app(device = ..)]
const APP: () = {
    static mut SHARED: u64 = 0;

    #[init(spawn = [foo])]
    fn init() {
        spawn.foo();
    }

    #[task(resources = [SHARED])]
    fn foo() {
        // this would have been restricted with `call = [baz]` (restriction #3)
        // .. do stuff with `resources.SHARED` ..

        // instead of `call.baz()`
        common(resources.SHARED);
    }

    #[task]
    fn bar() {
        spawn.baz();
    }

    #[task(resources = [SHARED])]
    fn baz() {
        common(resources.SHARED);
    }
};
fn common(shared: impl Resource<Data = u64>) {
    shared.lock(|shared| {
        *shared += 1;

        println!("{}", *shared);
    });
}
```

Because all resource dependencies need to be specified as function arguments no
restriction needs to be enforced at the DSL level -- Rust aliasing rules will be
enforced by the compiler.

TeXitoi, 2019-01-25:
As a user, this seems quite complicate to understand, when the alternative is quite natural.

Do you have any case when this feature is a big advantage?

japaric, 2019-01-25:
> As a user, this seems quite complicate to understand, when the alternative is quite natural.

I fully agree.

> Do you have any case when this feature is a big advantage?

I do not and would love to see one.

This is something that @perlindgren thought would be a good fit for the framework. I have merely made a design about it, but I'm personally not in favor of this RFC.

perlindgren, 2019-01-26:
Hi @TeXitoi  and all

The RFC is a step towards completeness to the original RTFM model, allowing both asynchronous task invocations (having no immediate return value) and synchronous invocations having an immediate return value and blocking behaviour (like we use to for an ordinary function call in Rust and most other languages.) The RFC allows a unified API to both types of task invocations. A task is different from a function in that it holds information about the resource utilization and dependencies to other tasks in the system (and priorities).

So what is the use case then? Code refactoring, and separation of concerns.

You as a programmer can factor out common functionality into sub-tasks and don't worry at call-site about the resource and communication behaviour of the factored out code. 

So what are the alternatives. 
1) implement something along the lines of the RFC.
2) encode it in standard Rust (as the RFC indicates possible).

Currently 2) is limited to factoring out code that does not spawn other tasks, so we might want/need to do something about this in any case.

For implementing 1), the analysis for cycles required for soundness is not overly complicated. Implementing 1) will offer a unified API to synchronous and asynchronous use, reducing the bloat of passing "meta" data regarding resource usage and communication behaviour. In particular we can re-use a task definition for both synchronous and asynchronous usage. This is not possible by 2.

There is still an open question here not covered by the RFC.

A task definition binds a Rust function to a task name, and in the signature a function may have a return value, meaningful in the synchronous setting while not in the asynchronous setting. (We are not async/await). We can choose to require tasks that are being used in an asynchronous fashion to have a *unit* return value, or we can allow this (and just ignoring the return value).

Another thing that the original RTFM model allowed was the abortion of queued while not already dispatched messages (asynchronous calls). The return value of `schedule` would then be a handler implementing `abort() -> Result<(),()>`. For this to be useful, such a handler should be possible to pass in a message (as a parameter) and/or store in a resource. This allows for flexible timeout handling and other use cases where a task should never dispatched/executed if its effect is no longer desired. (As a side effect the scheduling load of the system will be reduced.) Aborting a message that has already been dispatched would result in a `Err()` (as associated message is no longer in the timer queue). Maybe `abort` of scheduled message should be a separate RFC, however from a modelling perspective (and original RTFM implementation), dispatch of a message is a synchronous call of the corresponding task. Hence synchronous calls were brought up as an RFC first, as useful to the modelling of schedule/abort/dispatch.

/Per

korken89, 2019-02-14:
I agree that this seems to over-complicate the intended use and I would go with the alternative implementation using free Rust functions outside of RTFM.

Or keep it laying around until a strong use-case is available.

TeXitoi, 2019-02-14:
@perlindgren I know that's not so complicated to implement, but it really break readability: with this system, tasks can run to any priority. Understanding the relations between the priorities and the resources became very difficult for the user, and this is a very important relation that the user must be able to audit.

japaric, 2019-02-27:
Given that no strong use cases for this have been presented I'm going to propose we **postpone** this RFC. Let's vote on this and go into a 1-week FCP.

- [x] @japaric 
- [ ] @korken89 
- [ ] @TeXitoi

japaric, 2019-03-26:
The RTFM team has voted for postponing this and we have not received comments for a few weeks so I'm going to close this and label it as "postponed".

closed 2019-03-26 by japaric