rtic-rs/rtic #136: SysTick interrupt with disabled timer-queue

burrbull, 2019-02-08:
```
no variant named `SysTick` found for type `stm32f103xx::Interrupt` in the current scope
```
Same with stm32f1xx.

Sorry, when I've created #101 , I made mistake. I wanted to ask about SisTick interrupt, not about exception.

japaric, 2019-02-08:
SysTick is a Cortex-M *exception* available on all devices -- not a device specific *interrupt*. The reason why you are seeing that error is because there's no `SysTick` variant in [`stm32f103xx::Interrupt`](https://github.com/japaric/stm32f103xx/blob/master/src/lib.rs#L188-L309), this matches with the reference manual (see Section 10.1 Nested vectored interrupt controller (NVIC) in DocID13902 Rev 17)

burrbull, 2019-02-08:
I tried to run [reactive-blinky example](https://github.com/japaric/stm32f103xx-hal/blob/master/examples/reactive-blinky.rs.outdated) with #[interrupt(resources = [LED])] replaced with #[exception(resources = [LED])].

It was compiled fine. But when try to flash and run it, an error occurs.

japaric, 2019-02-08:
> an error occurs.

Could you give us more details? Do you get a panic / hard fault? Does the microcontroller hangs? something else?

burrbull, 2019-02-08:
A little later.

burrbull, 2019-02-08:
Strange. Compiled on other PC with rustc 1.34.0-nightly (f6fac4225 2019-02-03) and it works.
Sorry once again.

closed 2019-02-08 by burrbull