rtic-rs/rtic #143: [RFC] remove the difference between resources and late resources

TeXitoi, 2019-02-12:
# Summary

Resources and late resources are distinct in RTFM, when it's not really needed. static definition with SHOUTING_NAMES are also a bit strange when accessed by the `resources` magic parameter.

This RFC proposes a new syntax for declaring resources that remove the distinction between resources and late resources, and feel a bit more rustic.

# Background

Consider this example:

```rust
#[rtfm::app(device = ..)]
const APP: () = {
    // Strange syntax of late resource
    static mut FOO: ComplexThing = ();

    // As resources are presented as statics, they are in SHOUTING_SNAKE_CASE
    static mut BAR: i32 = 42;

    // `init::LateResource` is a magic type
    #[init]
    fn init() -> init::LateResources {
        init::LateResources {
            // strange SHOUTING_FIELD
            FOO: make_foo(),
        }
    }

    #[task(priority = 1, resources = [FOO, BAR])]
    fn foo() {
        // strange SHOUTING_FIELDS
        resources.BAR = resources.FOO.get_bar();
    }
};
```

# Design

The main idea is to declare a struct representing the resources. This struct is returned by init. Then, you have a partial view of this struct in the handlers.

```rust
#[rtfm::app(device = ..)]
const APP: () = {
    // define the resources by defining a struct
    #[resources]
    struct Resources {
        // as this is fields of a struct, we don't use shouting case here
        foo: ComplexThing,
        bar: i32,
    }

    // just return the declared Resources struct
    #[init]
    fn init() -> Resources {
        Resources {
            foo: make_foo(),
            // no more static initialization
            bar: 42,
        }
    }

    // declare the fieds of our `resources` variable as a subset of the Resource fields
    #[task(priority = 1, resources = { foo, bar })]
    fn foo() {
        // natural naming
        resources.bar = resources.foo.get_bar();
    }
};
```

# Drawbacks

`init` is no longer optional (when there is resources), thus complicating the simple case a bit.

As the `init` function returns all the resources, there is no more static initialisation of the static resources, potentially adding useless memcpy.

korken89, 2019-02-14:
I quite like this way of defining resources. The current way does look quite strange just to make proc-macros happy.
This approach also makes resources more explicit, something I value a lot as it reduces the chance for misinterpretation errors.

japaric, 2019-03-07:
TL;DR I'm in favor of this syntax but would prefer to only use it for late resources

> This approach also makes resources more explicit

I'd like to ask the participants to avoid using the word ["explicit"](https://boats.gitlab.io/blog/post/2017-12-27-things-explicit-is-not/) as it is a rather fuzzy term.

@korken89 Do you mean:

a. That the resources would be easier to *identify* since they wouldn't look like plain `static` variables? I think that resources are already easy to identify today: all `static` variables inside `#[app]` are resources (which makes them have one level of indentation) and the contents of `#[app]` are limited to a single file.

b. That the resources would be easier *locate* since they would all be grouped in a `struct`? I wonder if that would still be considered an advantage once we allow the contents of `#[app]` to be split across multiple files -- with this proposal you would *not* be able to split resources across files.

c. Or something else?

> The current way does look quite strange just to make proc-macros happy.

I can agree with this sentiment.

---

@texitoi

> init is no longer optional

`init` is mandatory today. You get an error if you don't have an `init` function. Or did you mean that the return type of `init` would no longer be optional?

> there is no more static initialisation of the static resources, potentially
> adding useless memcpy.

Today non-late resources are initialized with a single highly optimized memcpy before `main` / `init` is called. With this proposal every resource would get initialized with a not optimized memcpy (`ptr::write`). I think that the effect on initialization time will be negligible but the effect on program size may be not.

---

I thought that @texitoi was going to propose the `struct` syntax only for *late* resources, not for all resources. My preference would be to use the `struct` syntax only for late resources, for the following reasons:

a. Resources will initial values will continue to be initialized with an optimized memcpy, preserving today's performance and program size. IOW, this would be only a syntactic change.

b. In the future it would let users move non-late resources to other files.

Using the `struct` syntax for late resources means that all late resources must be declared in the same spot. I don't really see this as a limitation because, for readability, you'll likely place their declaration right next to `init`.

This is how I picture late resources in a struct integrating with a crate level `app` attribute:

``` rust
#![app]

#[late]
struct Resources {
    radio: Radio,
    // ..
}

#[init]
fn init() -> Resources {
    // ..
}

mod net {
    // owned resource
    #[resource] // clearly this is not a plain `static` variable
    static mut packet_counter: usize = 0;

    #[interrupt(binds = EXTI0, resources = [packet_counter, radio])]
    fn on_new_packet() {
        // ..
    }
}
```

korken89, 2019-03-12:
What I mean from my size is both `a` and `b`, they are both easier to identify and locate.

TeXitoi, 2019-03-12:
Yeah, I mean you have to returns the resources in init.

The point was to have the `resources` variable be a subset of your declared `Resources` struct. It feels more intuitive. Now, I know the drawback of these useless copies.