rtic-rs/rtic #144: Warnings with multi word resources

TeXitoi, 2019-02-12:
When I compile https://github.com/TeXitoi/rusty-clock with beta and nightly, I have these warnings:

```
$ cargo build --release
   Compiling rusty-clock v0.1.0 (/home/texitoi/dev/rusty-clock)
warning: type `ALARM_MANAGER` should have a camel case name
  --> src/main.rs:47:16
   |
47 |     static mut ALARM_MANAGER: alarm::AlarmManager = ();
   |                ^^^^^^^^^^^^^ help: convert the identifier to camel case: `AlarmManager`
   |
   = note: #[warn(non_camel_case_types)] on by default

warning: type `RTC_DEV` should have a camel case name
  --> src/main.rs:45:16
   |
45 |     static mut RTC_DEV: stm32f103xx_rtc::Rtc = ();
   |                ^^^^^^^ help: convert the identifier to camel case: `RtcDev`

warning: type `FULL_UPDATE` should have a camel case name
  --> src/main.rs:56:16
   |
56 |     static mut FULL_UPDATE: bool = false;
   |                ^^^^^^^^^^^ help: convert the identifier to camel case: `FullUpdate`

    Finished release [optimized + debuginfo] target(s) in 2.57s
```

No warning on stable.

I suppose the name of the resource is used to generate a type without silenting the warning.

Versions:
```
   stable-x86_64-unknown-linux-gnu unchanged - rustc 1.32.0 (9fda7c223 2019-01-16)
     beta-x86_64-unknown-linux-gnu unchanged - rustc 1.33.0-beta.6 (b203178b6 2019-02-05)
  nightly-x86_64-unknown-linux-gnu unchanged - rustc 1.34.0-nightly (57d7cfc3c 2019-02-11)
```

perlindgren, 2019-02-13:
Hi

I think there is already a PR fixing this…. (rustc now put harder restriction to proc macro generated code…)

Best
  Per

TeXitoi, 2019-02-13:
Note that there is much less warnings than in 0.4.0 so I think theses particular cases was forgotten as it need multi word resource name (mono word don't warn as full caps are considered proper PascalCase).

commit 87c69b68 (2019-02-13): Merge #145

145: fix non_camel_case_types warnings r=japaric a=japaric

closes #144
PR #145 (2019-02-13): fix non_camel_case_types warnings
closes #144
closed 2019-02-13 by bors[bot]