rtic-rs/rtic #146: [RFC] drop resources. prefix

japaric, 2019-02-13:
## Summary

Stop grouping resources in a `resources` struct and instead directly inject them
into the function namespace.

## Motivation

Reduce the amount of typing when using resources.

## Design

The DSL will stop creating the `$task::Resources` struct and instead of
injecting a `resources` value in the function namespace it will inject all the
declared resources. For example,

``` rust
// today you write
#[task(resources = [FOO, BAR])]
fn baz() {
    resources.FOO.lock(|foo| ..);

    // ..

    *resources.BAR += 1;
}

// with this RFC, that code becomes
#[task(resources = [FOO, BAR])]
fn baz() {
    FOO.lock(|foo| ..);

    // ..

    *BAR += 1;
}
```

## Drawbacks

### Readability

This change can negatively impact the readability of code. With the current
approach it's clear that the fields of `resources` match what's declared in
`resources = [..]`; it's also consistent with how `spawn` and `schedule` work.

With the proposed RFC it becomes less clear that you can only use resources
declared in `resources = [..]`. For example,

``` rust
static COUNT: AtomicUsize = AtomicUsize::new(0);

#[rtfm::app(..)]
const APP: () = {
    static mut FOO: u32 = 0;
    static mut BAR: u32 = 0;
    static mut BAZ: u32 = 0;

    #[task(resources = [FOO, BAR])]
    fn foo() {
        FOO.lock(|foo| ..); // OK

        // why doesn't this work
        BAZ.lock(|baz| ..); //~ ERROR value not found

        // but this works
        let curr = COUNT.fetch_add(1, Ordering::Relaxed);
    }
};
```

### Trampolines

Writing a trampoline function would require more work as the user would have to
either write their own `struct` to pass resources as a single argument, or make
their function take several arguments.

``` rust
// today
const APP: () = {
    #[task(resources = [FOO, BAR, BAZ, QUUX])]
    fn foo() {
        trampoline(resources);
    }
};

// maybe in another module
fn trampoline(resources: foo::Resources) {
    // ..
}
```

``` rust
// with this RFC
const APP: () = {
    #[task(resources = [FOO, BAR, BAZ, QUUX])]
    fn foo() {
        trampoline(Resources {
            FOO, BAR, BAZ, QUUX
        });
    }
};

struct Resources {
    FOO: resources::FOO,
    BAR: resources::FOO,
    BAZ: resources::FOO,
    QUUX: resources::FOO,
}

fn trampoline(resources: Resources) {
    // ..
}
```

``` rust
// with this RFC, another option
const APP: () = {
    #[task(resources = [FOO, BAR, BAZ, QUUX])]
    fn foo() {
        trampoline(FOO, BAR, BAZ, QUUX);
    }
};

fn trampoline(
    FOO: resources::FOO,
    BAR: resources::BAR,
    BAZ: resources::BAZ,
    QUUX: resources::QUUX,
) {
    // ..
}
```

### Shadowing

Directly adding all resources into the value namespace of a function increases
the chance of shadowing outer values, which makes fully qualified path necessary
for disambiguation.

``` rust
// today
static COUNT: AtomicUsize = AtomicUsize::new(0);

#[rtfm::app(..)]
const APP: () = {
    static mut COUNT: u32 = 0;

    #[task(resources = [COUNT])]
    fn foo() {
        resources.COUNT = COUNT.fetch_add(1, Ordering::Relaxed);
    }
};

// with this RFC
static COUNT: AtomicUsize = AtomicUsize::new(0);

#[rtfm::app(..)]
const APP: () = {
    static mut COUNT: u32 = 0;

    #[task(resources = [COUNT])]
    fn foo() {
        COUNT = crate::COUNT.fetch_add(1, Ordering::Relaxed);
    }
};
```

korken89, 2019-02-14:
I would not do this. Being explicit is very important IMO, and it makes it less ergonomic to use with external code.
The small improvement of not having to write `resources.` does not justify the drawbacks.

japaric, 2019-02-27:
I think people agree that the drawbacks outweigh the benefits of this proposal so I'm going to propose we **reject** this RFC. Let's vote on this and go into a 1-week FCP.

- [x] @japaric 
- [ ] @korken89 
- [ ] @TeXitoi

perlindgren, 2019-02-27:
Hi

As it is now its still a bit unclear where this resources parameter comes from, so I don't think readability is worse, but the name space issue (collision), and trampolining (passing resources forward) are for sure issues with this approach.

All in all, I like the clearness of RTFM3, with a task being called with a context collecting all called parameters.

Then the task signature follows that of Rust. I think this approach was discussed in another issue/thread, not sure it reached RFC stage.

So regarding #146, we are likely better off not changing/adding it now.

More importantly, is to investigate an idiomatic approach to writing drivers, how should we deal with callbacks (closures?), how should we deal with shared resources (calling into a driver with resources already claimed might not be the best option).

Another thing:

The original RTFM model had the notion of synchronous task invocations (like spawn but called directly). The same task definition can then be called either synchronously or in an async manner. (This sync/async pattern reflects the message passing model similar to actors which stems > 20 years back, prior to the async/await patterns later introduced, and is now trickling into Rust).

In the original RTFM the return value of  async is a handler to the message (allowing for abortion enqueued messages from the timer queue). This pattern may be useful to implementing timeouts, where the success path removes the timeout message. (One can think of a syntax for this).

Timeouts are quite common, and can be used for different purposes.

Best

  Per

japaric, 2019-03-26:
This team has voted for rejecting this RFC and no concerns have been raised for a few weeks so I'm going to close this and label it as "rejected".

closed 2019-03-26 by japaric

TeXitoi, 2019-03-26:
Sorry, I missed the call for vote.