rtic-rs/rtic #147: [RFC] All functions must be safe

japaric, 2019-02-13:
## Summary

Functions marked with any of these attributes: `exception`, `idle`, `init`,
`interrupt` and `task` must be safe `fn`s and not `unsafe fn`s.

## Motivation

Put us in line with rust-lang/rfcs#2585. All the functions inside the `app`
block can only be called by the hardware and are safe to call by the hardware
(the DSL ensures that's the case) -- this should be reflected in the signature.

If a user wants to use unsafe code inside any of these functions they should
explicitly use an `unsafe` block instead of using `unsafe fn` as a short-hand
for `unsafe fn() { unsafe { .. } }`.

## Design

The DSL will reject `unsafe fn`s. For example:

``` rust
#[rtfm::app(..)]
const APP: () = {
    #[init]
    fn init() { .. } // OK

    #[idle]
    unsafe fn idle() -> ! { .. } //~ ERROR: `idle` must have signature `fn() -> !`
};
```

## Drawbacks

Unsafe code will suffer from (even more) rightward drift.

``` rust
#[rtfm::app(..)]
const APP: () = {
    #[init]
    fn init() {
        unsafe {
            // unsafe stuff
        }

        // safe stuff
    }
};
```

TeXitoi, 2019-02-13:
That's almost a safety bug, as the framework have no way of knowing the contract to allow calling these function.

japaric, 2019-02-13:
> That's almost a safety bug

Can you elaborate on what you mean by "that"? Do you mean the current state or something else? (Today, the framework won't let call any of these functions from software; that's not going to change)

TeXitoi, 2019-02-13:
I mean if your interrupt handler is unsafe, RTFM has no way of knowing in which context this function can be called.

korken89, 2019-02-14:
I would say that this is a good addition, and to be in line with [rust-lang/rfcs#2585](https://github.com/rust-lang/rfcs/pull/2585), makes sense.

japaric, 2019-03-26:
I think this seems uncontroversial so let's vote for *approving* this RFC:

- [x] @japaric 
- [ ] @korken89 
- [ ] @TeXitoi

japaric, 2019-04-16:
This has enough approvals and no concerns have been raised over the course of several weeks so this RFC is now officially accepted. :tada:

commit bc024f19 (2019-05-01): Merge #176

176:  implement RFCs 147 and 155, fix #141, etc. r=japaric a=japaric

This PR:

- Implements RFC 147: "all functions must be safe"

- Implements RFC 155: "explicit Context parameter"

- Implements the pending breaking change #141: reject assign syntax in `init`
  (which was used to initialize late resources)

- Refactors code generation to make it more readable -- there are no more random
  identifiers in the output -- and align it with the book description of RTFM
  internals (see PR #175).

- Makes the framework hard depend on `core::mem::MaybeUninit` and thus will
  require nigh [... cut]
PR #176 (2019-05-01): implement RFCs 147 and 155, fix #141, etc.
This PR:

- Implements RFC 147: "all functions must be safe"

- Implements RFC 155: "explicit Context parameter"

- Implements the pending breaking change #141: reject assign syntax in `init`
  (which was used to initialize late resources)

- Refactors code generation to make it more readable -- there are no more random
  identifiers in the output -- and align it with the book description of RTFM
  internals (see PR #175).

- Makes the framework hard depend on `core::mem::MaybeUninit` and thus will
  require nightly until that API is stabilized.

- Fi [... cut]
closed 2019-05-01 by bors[bot]