rtic-rs/rtic #156: [RFC] link to real world examples from the book

japaric, 2019-02-19:
## Summary

Link to applications written by the community from the RTFM book.

## Motivation

The RTFM book covers the functionality and syntax of the framework but mostly
contains contrived examples. Readers have expressed their wish to see more
"real world" examples.

This RFC proposes that we link to such examples from the book but instead of
creating and maintaining them ourselves it proposes that we link to existing
examples built by the community.

## Design

Add a "Real World Examples" page at the end of the "RTFM by example"
chapter. In that page list applications using the following format:

``` markdown
- [$author/$project](https://example.org/$project/source/code). A few sentences
about the project.
```

Where the link to the source code must be a tag or snapshot, and not a branch
like `master` or `dev`.

The entries in this list will be submitted by the project authors as PRs. The
developers / maintainers of RTFM will evaluate the project and decide whether to
include it in the list or not.

These points will be considered when evaluating a project:

- The source code of the project must be public. The license doesn't matter as
  long as long as the code is public.

- The project must depend on a crates.io *release* of `cortex-m-rtfm`. For
  example, v0.4.2 is OK, but pre-release v0.5.0-beta.1 is not.

- Is the application using `unsafe` in a way that could compromise the safety
  guarantees provided by the framework?

- Is the code reasonably documented / commented?

If the list of examples is growing too large (a good problem to have!) the RTFM
devs may decide to stop accepting new entries.

The RTFM devs are free to remove examples that depend on EOL-ed releases of
`cortex-m-rtfm` from the list without notifying the authors.

liamim, 2019-02-20:
Are there _current_ real-world examples that you know of?

perlindgren, 2019-02-20:
Hi

It depends on what you define as real world. I run a course in design of embedded systems, since last year the students implement the embedded software using RTFM (RTFM2 last year, and this year on RTFM4).  As there are no pre-requisites to Rust experience (in fact some students take the course with very little programming background) their developed code serves more as a proof of concept, than well designed, polished, well documented code examples.

This year:
1) A diigital effect processor, interfacing a 4270 audio codec (over I2C/I2S), lcd display (over ISP). Hardware support to USB OTG.
2) A digital theremin, with IR based range detectors (ADC), serial (midii) over USB (using FTDI), audio out over internal PWM, hardware support for USB OTG.
… 4 more… including step/pulse meter, remote toaster control, electric lock with card reader, audio spectrum analyser.

Last year:
GBA cartige emulator without menu written in Rust, … 5 more …

Best
  Per

TeXitoi, 2019-02-20:
https://github.com/TeXitoi/rusty-clock

hdhoang, 2019-02-20:
we have https://github.com/ah-/anne-key, but the fact that it currently uses #87 is disqualification. I have just hooked up programmers to my keyboard in hope of porting it to the latest RTFM.

korken89, 2019-02-23:
Having link to real-world examples are good - however making sure that link follow the requirements as projects evolve will be a hassle.
Are we sure we want this extra administration?

TeXitoi, 2019-02-23:
That's why we would point to a commit, and can sometime make a bit of cleanups if the examples are out of date. It would be of the responsibility of the example authors to update the link when RTFM evolve.

japaric, 2019-02-26:
@korken89 what @TeXitoi said. If the authors don't update the tag / commit-id we'll eventually remove their outdated examples (e.g. GC all 0.4.x examples when we release v0.6.0). In the book we'd keep the examples grouped by minor version then GC-ing is just removing a whole group / section.

@hdhoang let me know if you run into trouble moving to v0.4.x, or if you need help!

hdhoang, 2019-02-27:
thanks for the great work and the offer!! we will be collecting our thoughts in ah-/anne-key#80, and maybe ping you over there once we're too stumped.

japaric, 2019-03-26:
This proposal has only received positive feedback so let's vote for *approving* this RFC.

- [x] @japaric 
- [ ] @korken89 
- [ ] @TeXitoi

japaric, 2019-04-16:
This has enough approvals and no concerns have been raised over the course of several weeks so this RFC is now officially accepted. :tada: 

People that have projects that use RTFM, you are now welcome to submit your projects to the book!

closed 2021-03-04 by korken89