rtic-rs/rtic #157: [RFC] ride the Rust trains

japaric, 2019-02-19:
## Summary

Publish *pre*-releases of new minor versions, e.g. v0.5.0-beta.1, that target
the beta channel of the Rust compiler. When the beta release of the compiler
makes it way into stable publish the final release of the new minor version,
e.g. v0.5.0.

## Motivation

The framework currently depends on (or works around) unstable core APIs like
`MaybeUninit` and unstable language features like `const_fn`, but thankfully these
features will soon make their way into stable.

We'd like to make the benefits of internally using these soon-to-be-stabilized
APIs / features to our stable users as soon as possible, but at the same time
we'd like to have some time to test these changes before publishing a new minor
version. 

Doing pre-releases that target beta hits the sweet spot between these two
requirements. It gives us a 6 week test cycle while at the same time letting
stable users evaluate the next minor version of RTFM without switching to
the nightly channel.

## Design

Let's illustrate the proposed timeline with an example.

Assume that `MaybeUninit` will be stabilized by 1.35. This means that the API
will *not* require a feature gate by 1.35-beta, which is scheduled for release
on April 12.

The idea would be to *pre*-release RTFM v0.5.0-alpha, which internally uses
`MaybeUninit` and requires rustc 1.35-beta, on that date or slightly later and
then have a 3-week *alpha* period. During the alpha period we are allowed to
make breaking changes to the API of RTFM, e.g. v0.5.0-alpha.2 changes an API
introduced in v0.5.0-alpha.1.

Once the alpha period ends we'll publish v0.5.0-beta.1 and start a 3-week beta
period where no breaking changes can occur, e.g. v0.5.0-beta.2 can fix bugs but
not change the API exposed by v0.5.0-beta.1.

When Rust 1.35 is released on May 24, or slightly later, we'll release v0.5.0 of
RTFM and yank all the alpha and beta pre-releases.

The table below shows the full timeline for our example

| Date     | Stable | Beta | RTFM        |
|----------|--------|------|-------------|
| March 01 |   1.33 | 1.34 |             |
| April 12 |   1.34 | 1.35 | 0.5.0.alpha |
| May 3    |   1.34 | 1.35 | 0.5.0-beta  |
| May 24   |   1.35 | 1.36 | 0.5.0       |

These are meant to be guidelines; the RTFM devs may decide to tweak the
timeline, for example by shrinking the alpha period to 2 weeks or extending it
to 4 weeks.

korken89, 2019-02-23:
Overall I think this is good, but one should also consider the extra workload - are you ok with this?
IMO RTFM was taken out of nightly-only a bit early, but doing this might be the best of both worlds?

japaric, 2019-04-16:
Update: Rust 1.34 and 1.35-beta were released last week but stable `MaybeUninit` didn't make it into the beta cut-off. Let's hope that it makes it into 1.36-beta!

We have a few RFCs approved that need a minor version bump. Perhaps we should consider releasing an early alpha that targets nightly as soon as `MaybeUninit` doesn't require a feature gate?

korken89, 2019-04-28:
I am strongly for this, lets have an early alpha on nightly while we await stabilization

japaric, 2019-05-02:
Update: MaybeUninit has been formally proposed for stabilization in rust-lang/rust#60445. Hopefully it will make it into 1.36-beta (due in 3 weeks).

korken89, 2019-05-02:
:tada:

timbod7, 2019-06-16:
I've a bit confused by the versioning. If I'm planning to use cortex-m-rtfm for a new project should I used current master (which appears to be the work in progress for v0.5) or should I use the published 0.4.3? 

I thing 0.4.3 will work with the current rust stable toolchain, whereas master needs nighty? On the other hand the published documentation book (https://japaric.github.io/rtfm5/book/en/preface.html) is for a future 0.5 release?