rtic-rs/rtic #164: `impl Trait` in resources?

little-arhat, 2019-03-17:
Is it possible to use `impl Trait` in resources?

I get following error on nightly, even with corresponding feature enabled. 

```
error[E0562]: `impl Trait` not allowed outside of bindings or function and inherent method return types
  --> src/main.rs:68:19
   |
68 |     static mut B: impl X = ();
   |                   ^^^^^^
```

little-arhat, 2019-03-17:
Without this feature you have to provide type aliases for complex types, and you also can't use opaque types as resources.

little-arhat, 2019-03-17:
I think I've got it to work using existentials, but I wonder it this the best way:

```
mod c {
    use super::hprintln;
    pub existential type C: T;
    pub trait T {
        fn dodo(&self);
    }
    pub struct TT;
    impl T for TT {
        fn dodo(&self) {
            hprintln!("xx").unwrap();
        }
    }

    pub fn mk() -> C {
        TT
    }
}
use c::T;

#[app(device = stm32f30x)]
const APP: () = {
    static mut EXTI: stm32f30x::EXTI = ();
    static mut MPU: MPU9250 = ();
    static mut I: Instant = ();
    static mut CC: CyclesToTime = ();
    static mut ExT: c::C = ();
// rest is ommitted
}
```

https://gist.github.com/1aefa621a9e179f267afe532a4ddfc9e

perlindgren, 2019-03-17:
Hi Roma

Not sure exactly what you want to do.

This pattern (tried in Playground not RTFM) would allow you to store a reference to an implementer of X in B. The type that implements X is not mentioned here, could be anything (opaque if so wished I believe).

trait X {}
static mut B:Option<&'static X> = None;

or…
mod c {
    pub trait X { }

    struct C {
        private_field: u32
    }

    impl X for C {}

    fn mk_c() -> C {
        C { private_field: 0}
    }
}

static mut C:Option<&'static c::X> = None;

if you want some scoping and opaque data type.

For more on visibility see, https://doc.rust-lang.org/reference/visibility-and-privacy.html.

… or maybe I got your use case wrong…

Best regards,
  Per

To learn more, run the command again with --verbose.

perlindgren, 2019-03-17:
Hi Roma..

Is trait sealing perhaps what your after.

https://rust-lang-nursery.github.io/api-guidelines/future-proofing.html

Restricting the implementation of a trait, to some specific scope. Else associated items in a pub trait are pub by default.

Best regards,
  Per
On 17 Mar 2019, at 19:24, Per Lindgren <per.lindgren@ltu.se<mailto:per.lindgren@ltu.se>> wrote:

Hi Roma

Not sure exactly what you want to do.

This pattern (tried in Playground not RTFM) would allow you to store a reference to an implementer of X in B. The type that implements X is not mentioned here, could be anything (opaque if so wished I believe).

trait X {}
static mut B:Option<&'static X> = None;

or…
mod c {
    pub trait X { }

    struct C {
        private_field: u32
    }

    impl X for C {}

    fn mk_c() -> C {
        C { private_field: 0}
    }
}

static mut C:Option<&'static c::X> = None;

if you want some scoping and opaque data type.

For more on visibility see, https://doc.rust-lang.org/reference/visibility-and-privacy.html.

… or maybe I got your use case wrong…

Best regards,
  Per

To learn more, run the command again with --verbose.

little-arhat, 2019-03-17:
@perlindgren, no, I'm not interested in sealed traits, I just want to define resource that has quite a long type. Users (tasks) shouldn't have to know those details, they should only its trait.

I guess I can go with references, but then it's another level of indirection, and I would have to store my object as a static manually, without using rtfm resources. 

type: 
```mixer::MotorCtrl<nalgebra::base::matrix::Matrix<f32, nalgebra::base::dimension::U6, nalgebra::base::dimension::U4, nalgebra::base::array_storage::ArrayStorage<f32, nalgebra::base::dimension::U6, nalgebra::base::dimension::U4>>, (alt_stm32f30x_hal::pwm::PwmBinding<alt_stm32f30x_hal::gpio::PA0<alt_stm32f30x_hal::gpio::PullUp, alt_stm32f30x_hal::gpio::AltFn<alt_stm32f30x_hal::gpio::AF1, alt_stm32f30x_hal::gpio::PushPull, alt_stm32f30x_hal::gpio::MediumSpeed>>, alt_stm32f30x_hal::timer::tim2::Channel<alt_stm32f30x_hal::timer::CH1, alt_stm32f30x_hal::timer::Pwm1>, alt_stm32f30x_hal::gpio::AF1>, alt_stm32f30x_hal::pwm::PwmBinding<alt_stm32f30x_hal::gpio::PA1<alt_stm32f30x_hal::gpio::PullUp, alt_stm32f30x_hal::gpio::AltFn<alt_stm32f30x_hal::gpio::AF1, alt_stm32f30x_hal::gpio::PushPull, alt_stm32f30x_hal::gpio::MediumSpeed>>, alt_stm32f30x_hal::timer::tim2::Channel<alt_stm32f30x_hal::timer::CH2, alt_stm32f30x_hal::timer::Pwm1>, alt_stm32f30x_hal::gpio::AF1>, alt_stm32f30x_hal::pwm::PwmBinding<alt_stm32f30x_hal::gpio::PA2<alt_stm32f30x_hal::gpio::PullUp, alt_stm32f30x_hal::gpio::AltFn<alt_stm32f30x_hal::gpio::AF1, alt_stm32f30x_hal::gpio::PushPull, alt_stm32f30x_hal::gpio::MediumSpeed>>, alt_stm32f30x_hal::timer::tim2::Channel<alt_stm32f30x_hal::timer::CH3, alt_stm32f30x_hal::timer::Pwm1>, alt_stm32f30x_hal::gpio::AF1>, alt_stm32f30x_hal::pwm::PwmBinding<alt_stm32f30x_hal::gpio::PA3<alt_stm32f30x_hal::gpio::PullUp, alt_stm32f30x_hal::gpio::AltFn<alt_stm32f30x_hal::gpio::AF1, alt_stm32f30x_hal::gpio::PushPull, alt_stm32f30x_hal::gpio::MediumSpeed>>, alt_stm32f30x_hal::timer::tim2::Channel<alt_stm32f30x_hal::timer::CH4, alt_stm32f30x_hal::timer::Pwm1>, alt_stm32f30x_hal::gpio::AF1>, alt_stm32f30x_hal::pwm::PwmBinding<alt_stm32f30x_hal::gpio::PA6<alt_stm32f30x_hal::gpio::PullUp, alt_stm32f30x_hal::gpio::AltFn<alt_stm32f30x_hal::gpio::AF2, alt_stm32f30x_hal::gpio::PushPull, alt_stm32f30x_hal::gpio::MediumSpeed>>, alt_stm32f30x_hal::timer::tim3::Channel<alt_stm32f30x_hal::timer::CH1, alt_stm32f30x_hal::timer::Pwm1>, alt_stm32f30x_hal::gpio::AF2>, alt_stm32f30x_hal::pwm::PwmBinding<alt_stm32f30x_hal::gpio::PA7<alt_stm32f30x_hal::gpio::PullUp, alt_stm32f30x_hal::gpio::AltFn<alt_stm32f30x_hal::gpio::AF2, alt_stm32f30x_hal::gpio::PushPull, alt_stm32f30x_hal::gpio::MediumSpeed>>, alt_stm32f30x_hal::timer::tim3::Channel<alt_stm32f30x_hal::timer::CH2, alt_stm32f30x_hal::timer::Pwm1>, alt_stm32f30x_hal::gpio::AF2>)>```

perlindgren, 2019-03-17:
Hi Roma

Hmm, if you store them outside of RTFM they would require unsafe access. That's the main point with RTFM, safe access to shared resources. If only one task can access/mutate the resource, then you can make it a task local (with mutable safe access without any need for locking).

The same rules to storing trait objects should apply with or without RTFM, so you would gain nothing by unsafe access.

Let us know how you ended up solving the problem and/or if you run into further problems/questions. (There might still be things in RTFM that should be possible to express, but the current implementation would block you, and we are eager to work those out...)

Best
  Per

TeXitoi, 2019-03-17:
Existensial is the solution: RTFM need a type for its code génération, and impl ils not a type.

little-arhat, 2019-03-17:
@TeXitoi however, with existentials I still can't hide type parameters, due to `part of concrete type but not used in parameter list for existential type`, so I still have to expose internal details and create huge type aliases. 

Or maybe I'm missing something?

TeXitoi, 2019-03-17:
I'm not really award of thé existential feature. I think you may use something like existential un trait. Can you give your needs, I might try to find something tomorrow.

japaric, 2019-03-19:
@little-arhat what you *really* want is whole-crate type inference. That is you want this to compile:

``` rust
static mut X: Option<_> = None;

fn main() {
    unsafe {
        X = Some(0i32);
    }
}
```

Then you don't need any intermediate trait. However, I haven't seen any RFC for that.

Existential types and `impl Trait` won't really help. AFAICT, the opaque / existential type is only resolved when it appears in the signature of a function which defeats the point as you would need to a dummy function that calls a constructor or similar (the `c::mk` in your example).

> however, with existentials I still can't hide type parameters

I imagine you are trying to write something like this:

``` rust
#![feature(existential_type)]

use core::ops::{Deref, DerefMut};

existential type Foo: DerefMut;

static mut X: Option<Foo> = None;

fn foo() -> i32 {
    unsafe { *X.as_ref().unwrap().deref() + 1 }
}

fn main() {
    unsafe {
        X = Some(0i32); // <- `Foo = i32`
    }
}
```

AIUI this won't work because of how `rustc` does type unification today. Whether this would work some day I don't know; you would need to ask the compiler team.

AFAICS we can't do anything about this on the RTFM side as it needs better compiler support.

P.S. I can relate to the inconvenience of the long types. I get rid of them using [type aliases](https://github.com/japaric/jnet/blob/b5bd70cb998b1e01236f6b07ebc516a0359fde3d/firmware/src/lib.rs#L40) in a library crate that the application links to.

little-arhat, 2019-03-19:
@japaric thanks for info and insights! Yeah, I guess that's not strictly related to `cortex-m-rtfm`, but more of a general rust question...

Yes, whole-crate inference would help, but I guess we'll not get it (anytime soon) %).

But I also wonder how to create _abstract_ types, in a sense that they should hide internal and unintresting details: users of a library should know that `lib::Thing` can do `SomeThings`, but they don't have to know that `Thing` is parametrized over N other types. I guess type aliases could help a bit, but then in the type error you'll still see `nalgebra::base::matrix::Matrix<f32, nalgebra::base::dimension::U6, nalgebra::base::dimension::U4, nalgebra::base::array_storage::ArrayStorage<f32, nalgebra::base::dimension::U6, nalgebra::base::dimension::U4>>` instead of `Matrix6` that you've used.

Adding dynamic dispatch in the mix would also help, but for a price.

Another related thing bothers me: I have a sensor that my program communicates with over `I2c`. I've defined `type I2c = device::I2c<...>` with all the pins and details,  and I use `Sensor<I2c>` as a type of resource. Then, I want to use another i2c connected sensors and I use something like [shared-bus](https://github.com/Rahix/shared-bus) to make it work:

```
    let i2c = ...
    let bus = SharedBus::new(i2c);
    let mut sensor1 = Sensor1::new(bus.acquire());
    let mut sensor2 = Sensor2::new(bus.acquire());
```

(I acknowledge shared bus might not be ideal soluation in the RTFM, but bear with me).

But now the type of the sensor is something like `shared_bus::CortexMBus<shared_bus::CortexMProxy<RecCell<I2c>>, I2c>` (it's actually worse)! Of course, type aliases would help, especially if provided by library author, but is this really how it should be? 

Thanks for your input, everyone!

korken89, 2019-12-29:
This seems to be done discussing.

closed 2019-12-29 by korken89