rtic-rs/rtic #165: Share `core.SCB` with interrupts?

azasypkin, 2019-03-23:
Hey,

I tried to migrate to `rtfm` the other day and the only issue that I couldn't solve is sharing `core.SCB` with interrupts. If I'm not mistaken only `init` has access to mutable reference to `core.SCB` and that's it. 

My device is supposed to process USB/RTC interrupts and as a result it may want to toggle deep sleep mode (`SCB.scr` register) or do a soft reset (`SCB.aircr` register).

Am I missing something or there is no way currently to achieve that with RTFM?

hdhoang, 2019-03-24:
we have the same need for soft reset, and have made rust-embedded/cortex-m#138 to trigger it without having an instance of SCB. It's safe because it is one write, I don't know if deep sleep works similarly. I hope we can find a satisfactory solution.

azasypkin, 2019-03-24:
Ah, interesting, thanks for sharing @hdhoang! Yeah, I think same could work for [deep sleep](https://docs.rs/cortex-m/0.6.0/src/cortex_m/peripheral/scb.rs.html#585-589). In case we don't have other choice and your PR gets accepted I'll prepare the same for deep sleep.

hdhoang, 2019-05-06:
hi @azasypkin, you can start preparing a similar PR for deepsleep now, to enter the same deprecation cycle (0.6.1 to 0.7).

azasypkin, 2019-05-07:
Yep, thanks for the ping @hdhoang! Created PR: https://github.com/rust-embedded/cortex-m/pull/145

japaric, 2019-09-15:
This can be done in v0.5.x (not yet released but you can try master)

closed 2019-09-15 by japaric