rtic-rs/rtic #166: Panic when running on STM32F103RCT6

skammer, 2019-03-26:
I have trouble starting RTFM on STM32F103RCT6. It panics before entering init() function. Same code works OK on STM32F103C8T6 (blue pill).

Any tips how to troubleshoot this? I tried stepping through with gdb and it seems to panic sometime after calling `get_mut`, but I don't think it's enough information to diagnose what is happening.

```
#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

extern crate cortex_m;
extern crate cortex_m_rt as rt;
extern crate stm32f1xx_hal as hal;
extern crate cortex_m_semihosting;
extern crate panic_semihosting;
extern crate rtfm;

use cortex_m_semihosting::{hprintln};
use rtfm::app;

#[app(device = stm32f1xx_hal::stm32)]
const APP: () = {
    #[init]
    fn init() {
        hprintln!("init(baseline = {:?})", start).unwrap();
    }

    extern "C" {
        fn USART1();
    }
};
```

TeXitoi, 2019-03-27:
Can you minimize your code even more? at which step does it works? Can you run a non rtfm binary on the RCT6?

What about memory.x? Does setting the RAM and FLASH length to the real values help?

skammer, 2019-03-27:
I minimised the code and it seems to work when I run this:
```
#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

extern crate cortex_m;
extern crate cortex_m_rt as rt;
extern crate stm32f1xx_hal as hal;
extern crate cortex_m_semihosting;
extern crate panic_semihosting;
extern crate rtfm;

use rtfm::app;

#[app(device = stm32f1xx_hal::stm32)]
const APP: () = {
    #[init]
    fn init() {
        loop {
        }
    }
};
```

So the culprit appears to be related to `hprintln!` macro, as this code throws exception:
```
#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

extern crate cortex_m;
extern crate cortex_m_rt as rt;
extern crate stm32f1xx_hal as hal;
extern crate cortex_m_semihosting;
extern crate panic_semihosting;
extern crate rtfm;

use cortex_m_semihosting::{hprintln};
use rtfm::app;

#[app(device = stm32f1xx_hal::stm32)]
const APP: () = {
    #[init]
    fn init() {
        hprintln!("init(baseline = {:?})", start).unwrap();
        loop {
        }
    }
};
```

And it seems to fail when calling `hstdout()`
https://github.com/rust-embedded/cortex-m-semihosting/blob/5b99e18c1601702aed50e05f1208a06738730222/src/hio.rs#L53

And more specifically here:
https://github.com/rust-embedded/cortex-m-semihosting/blob/5b99e18c1601702aed50e05f1208a06738730222/src/hio.rs#L58

Right after calling `core::slice::<impl [T]>::as_ptr (self=...)`

I tried stepping through with gdb and here are the last instructions it ran:

[image]

Disassembling the binary with Hopper shows the following code a that location:

[image]

Before disassembling, I've set the base address to `0x8000000` to match the settings in linker script:

```
MEMORY
{
  FLASH (rx) : ORIGIN = 0x08000000, LENGTH = 256K
  RAM  (rwx) : ORIGIN = 0x20000000, LENGTH = 48K
}
```

This is what Cube generates for that specific MCU, so I presume the values are correct.

I did try running the code without rtfm and it works and blinks the led as expected:
```
// #![deny(unsafe_code)]
// #![deny(warnings)]
#![no_main]
#![no_std]

#[allow(unused)]

extern crate stm32f1xx_hal;
extern crate cortex_m_rt;
extern crate cortex_m_semihosting;
extern crate panic_semihosting;
extern crate cortex_m;
extern crate nb;

use stm32f1xx_hal::{
    prelude::*,
    pac,
    rtc::Rtc,
};

use nb::block;
use cortex_m_rt::entry;

#[entry]
fn main() -> ! {
    let dp = pac::Peripherals::take().unwrap();

    let mut pwr = dp.PWR;
    let mut rcc = dp.RCC.constrain();

    // Set up the GPIO pin
    let mut gpioc = dp.GPIOC.split(&mut rcc.apb2);
    let mut led = gpioc.pc13.into_push_pull_output(&mut gpioc.crh);

    // Set up the RTC
    // Enable writes to the backup domain
    let mut backup_domain = rcc.bkp.constrain(dp.BKP, &mut rcc.apb1, &mut pwr);
    // Start the RTC
    let mut rtc = Rtc::rtc(dp.RTC, &mut backup_domain);

    let mut led_on = false;
    loop {
        // Set the current time to 0
        rtc.set_seconds(0);
        // Trigger the alarm in 5 seconds
        rtc.set_alarm(5);
        block!(rtc.wait_alarm()).unwrap();
        if led_on {
            led.set_low();
            led_on = false;
        }
        else {
            led.set_high();
            led_on = true;
        }
    }
}
```

When I added `hprintln!("init").unwrap();` to the code above, it fails again during output, but this time during `cortex_m_semihosting::hio::write_all` while getting `butter.len()`
https://github.com/rust-embedded/cortex-m-semihosting/blob/5b99e18c1601702aed50e05f1208a06738730222/src/hio.rs#L67

May or may not be related to this issue: https://github.com/rust-embedded/cortex-m-semihosting/issues/24

perlindgren, 2019-03-27:
Hi

Did you try setting the memory.x to the default for the bluepill. (20k ram/64kb flash I think).

Best

  Per

https://github.com/rust-embedded/cortex-m-semihosting/blob/5b99e18c1601702aed50e05f1208a06738730222/src/hio.rs#L53

https://github.com/rust-embedded/cortex-m-semihosting/blob/5b99e18c1601702aed50e05f1208a06738730222/src/hio.rs#L58

[2019-03-27_18-14-04]<https://user-images.githubusercontent.com/17655/55088399-b9a23580-50bc-11e9-972b-03c6790695ae.png>

[2019-03-27_18-13-46]<https://user-images.githubusercontent.com/17655/55088506-e3f3f300-50bc-11e9-9638-88908385a383.png>

https://github.com/rust-embedded/cortex-m-semihosting/blob/5b99e18c1601702aed50e05f1208a06738730222/src/hio.rs#L67

May or may not be related to this issue: rust-embedded/cortex-m-semihosting#24<https://github.com/rust-embedded/cortex-m-semihosting/issues/24>

skammer, 2019-03-27:
Just tried. No change at all.

skammer, 2019-03-27:
Case closed, guys.

I have this board https://www.waveshare.com/Core103R.htm and it has _a switch_ instead of regular boot select jumpers.

Apparently, I switched it away from `FLASH` setting while handling the board and didn't notice until now.

Only one question left. Can we somehow detect that boot select is the wrong way around and warn people instead of silently failing?

TeXitoi, 2019-03-28:
For your last question, I don't know. But that's not the place for that, feel free to open an issue in cortex-m-semihosting.

I close this issue as it is fixed for you.

closed 2019-03-28 by TeXitoi