rtic-rs/rtic #168: voluntarily yield?

riaqn, 2019-04-15:
I'm quite new to Rust so I could be wrong, but is there some voluntarily yielding I can do? That is, the task is still runnable, but put at the back of the runnable queue.

japaric, 2019-04-21:
RTFM tasks have "run to completion" semantics -- they are *not* time-sliced threads like you normally see in OSes like Linux. So, no; you can't yield, or suspend, a task. You can only `return` from it and have something else (an interrupt or another task) invoke it (`spawn` / `schedule`) again -- tasks are event driven after all.

That being said you can implement cooperative multitasking which is closer to "straight-line threaded code" in `idle` (which is *not* a task and runs forever / never returns) using generators, futures, or what have you. Any such implementation needs no additional support from RTFM and there are many possible implementations with different trade offs so we leave end users go at it in whichever way they want (i.e. *we* provide no particular implementation in this crate).

riaqn, 2019-04-26:
Thanks @japaric for the explanation.

I found this: https://github.com/japaric/no-std-async-experiments

It looks very promising - may I ask when there will be a release?

japaric, 2019-05-02:
@riaqn I have no plans (time) to work on that any time soon.

If you have no more questions related to RTFM could you please close this issue?

riaqn, 2019-05-02:
Thanks. I'm closing the  issue.

closed 2019-05-02 by riaqn

Erhannis, 2024-10-24:
Did this change at some point?  I have two tasks that look like
```
  #[task(priority = 1)]
  async fn task1(_cx: task1::Context) {
      defmt::info!("Hello from task1!");

      loop {
          defmt::info!("1");
          Mono::delay(500_u64.millis()).await;
      }
  }

  #[task(priority = 1)]
  async fn task2(_cx: task2::Context) {
      defmt::info!("Hello from task2!");

      loop {
          defmt::info!("2");
          Mono::delay(1000_u64.millis()).await;
      }
  }
```

Despite never returning, they interleave as I'd expect, which suggests that they can yield execution to other tasks and then resume.  Which brings me to the question I was initially looking for - can you yield execution without specifying e.g. a delay or other condition, just "let other tasks run, then come back"?

(More confusingly, a loop I forgot in `idle` that only iterated a variable and occasionally printed _also_ kept running, and I haven't the foggiest how it'd do that without either using multiple cores or proper threads.  Only `idle` seems to be capable of arbitrary interruption, though; other tasks just hog execution.)

Skypr, 2025-06-10:
Hi @Erhannis , I was wondering the same.

I have come up with the following:

```RUST
async fn yield_task() {
    let mut yielded = false;
    core::future::poll_fn(|cx| {
        if yielded {
            return core::task::Poll::Ready(())
        } else {
            yielded = true;
            cx.waker().wake_by_ref();
            return core::task::Poll::Pending;
        }
    }).await;
}
```

which leads to proper interleaving of the tasks at the same priority.

Keep in mind that this will probably still have some performance impact.

Regarding idle: 
It is my understanding that this is the point of it. If no other task wants to run, the idle task will run. Which also means that this "yielding" is probably not really required, as you can just interleave the function calls in the idle task.

I just wanted to see if I could achieve this outcome.